inivation/caer #3: Visualizer: display custom value (text) and show as a bar (float), eg. Evts/sec

breznak, 2016-05-13:
Details: https://sourceforge.net/p/jaer/discussion/631959/thread/5a1a2c3c/ 

I'd like to visualize custom data along with the visualizations from `modules/visualizer/visualizer.h`.
For simplicity take I'd like to show a text `MyValue: val` and a bar (the val is a float on defined range).

> "...Now the Visualizer also supports custom renderer and event handler call-backs, so you can do your own drawing if needed."

An example is displaying Evts/second from statistics.

breznak, 2016-05-13:
@llongi would you accept this as a generic feature/example for the `Visualizer`, or should it be a custom private code?

llongi, 2016-05-30:
Statistics is automatically tracked and supported in all renderers.
But yeah, the idea is good, so I'd accept a good implementation of it.
Just not entirely sure how you'd defined the range in a generic way. It could take Point1Ds as input, then determine the max() for that packet, and draw them, or keep a running max(), or have some user passed configuration. I should probably add support for passing a user-controlled pointer into the renderer anyway, so that people can pass in or change parameters.

breznak, 2016-05-31:
> Just not entirely sure how you'd defined the range in a generic way. It could take Point1Ds as input, then determine the max() for that packet, and draw them, or keep a running max(), or have some user passed configuration. 

I'd agree on a config param set by the user, and just print a warning if the values overflow.