inivation/caer #6: Consider setting more information during the module's init()

breznak, 2016-05-31:
for example sending a "handshake" packet that would verify the connection works, and send static information about the hardware (e.g. `dvsSizeX`,..) 

Please see https://github.com/inilabs/caer/pull/2#discussion_r65140493 for context. 

This change would remove excessive checks from the `module.run()` function. Is it possible with the data-drive design? Would you consider it? 

(I'm aware it's done (workarounded) differently in the current code, but the proposed solution seems cleaner to me)

llongi, 2017-04-30:
Now that modules exist and we know the full connectivity at start of the execution loop, this can be done. A module will be able to ask "who do I get my data from?", and then wait for that to be ready (ie. simply return false from Init() until the source has started and setup its information, if it needs it), and then ask for information like size etc. from the sourceInfo node. Just an update that this is happening.

llongi, 2017-05-31:
Ok this has been successfully implemented in the modules branch, for the modules currently ported.
In Init() you can now use caerMainloopGetModuleInputIDs() to get the SourceIDs for the modules that will deliver data to that module, and you can then do caerMainloopGetSourceInfo(sourceID) in Init() to get the information needed. caerMainloopGetSourceInfo() now returns NULL if the source module is not running or not ready, so you can just block initialization until the source actually exists and has information that you need. The whole thing looks like this:

```
static bool Init(caerModuleData moduleData) {
	// Wait for input to be ready. All inputs, once they are up and running, will
	// have a valid sourceInfo node to query, especially if dealing with data.
	int16_t *inputs = caerMainloopGetModuleInputIDs(moduleData->moduleID, NULL);
	if (inputs == NULL) {
		return (false);
	}

	int16_t sourceID = inputs[0];
	free(inputs);

	// Get info from sourceInfo.
	sshsNode sourceInfo = caerMainloopGetSourceInfo(sourceID);
	if (sourceInfo == NULL) {
		return (false);
	}

	int16_t sizeX = sshsNodeGetShort(sourceInfo, "polaritySizeX");
	int16_t sizeY = sshsNodeGetShort(sourceInfo, "polaritySizeY");

	...
}
```

closed 2017-05-31 by llongi