inivation/caer #11: FileInput module broken:

federicohyo, 2016-06-10:
Recording and saving file seems to be broken with the error 10-FileInput: Failed to parse event packets.

./caer-bin 
2016-06-10 15:50:23 (TZ+0200): NOTICE: Logger: Initialization successful with log-level 5.
2016-06-10 15:50:23 (TZ+0200): NOTICE: Config Server: Ready and listening on 127.0.0.1:4040.
2016-06-10 15:50:23 (TZ+0200): ERROR: 10-FileInput: Failed to parse event packets.
Total events/second:          0 - Valid events/second:          02016-06-10 15:50:25 (TZ+0200): ERROR: 10-FileInput: Failed to parse event packets.

llongi, 2016-06-10:
No clue. Post your caer-config.xml please.

federicohyo, 2016-06-11:
just updated.. recording was done yesterday:

Federicos-MacBook-Air:caer_git federicocorradi$ ./caer-bin 
2016-06-11 12:10:37 (TZ+0200): NOTICE: Logger: Initialization successful with log-level 5.
2016-06-11 12:10:37 (TZ+0200): NOTICE: Mainloop: LASER SCANNER ENABLED
2016-06-11 12:10:37 (TZ+0200): NOTICE: Config Server: Ready and listening on 127.0.0.1:4040.
2016-06-11 12:10:37 (TZ+0200): ERROR: 10-FileInput: No compliant Format header found. Invalid file.
2016-06-11 12:10:37 (TZ+0200): ERROR: 10-FileInput: Failed to parse header.
2016-06-11 12:10:37 (TZ+0200): NOTICE: PoseCalibration CXX loadCalibrationFile(): Camera matrix and distorsion coefficients succesfully loaded
Total events/second:          0 - Valid events/second:          02016-06-11 12:10:38 (TZ+0200): NOTICE: PoseCalibration CXX loadCalibrationFile(): Camera matrix and distorsion coefficients succesfully loaded
Total events/second:          0 - Valid events/second:          02016-06-11 12:10:40 (TZ+0200): ERROR: 10-FileInput: No compliant Format header found. Invalid file.
2016-06-11 12:10:40 (TZ+0200): ERROR: 10-FileInput: Failed to parse header.

<sshs version="1.0">
    <node name="" path="/">
        <node name="1" path="/1/">
            <node name="1-DAVISFX3" path="/1/1-DAVISFX3/">
                <attr key="autoRestart" type="bool">true</attr>
                <attr key="busNumber" type="short">0</attr>
                <attr key="devAddress" type="short">0</attr>
                <attr key="runAtStartup" type="bool">true</attr>
                <attr key="serialNumber" type="string"></attr>
                <node name="DAVIS346B" path="/1/1-DAVISFX3/DAVIS346B/">
                    <node name="aps" path="/1/1-DAVISFX3/DAVIS346B/aps/">
                        <attr key="ADCTestMode" type="bool">false</attr>
                        <attr key="ColumnSettle" type="short">60</attr>
                        <attr key="EndColumn0" type="short">345</attr>
                        <attr key="EndColumn1" type="short">346</attr>
                        <attr key="EndColumn2" type="short">346</attr>
                        <attr key="EndColumn3" type="short">346</attr>
                        <attr key="EndRow0" type="short">259</attr>
                        <attr key="EndRow1" type="short">260</attr>
                        <attr key="EndRow2" type="short">260</attr>
                        <attr key="EndRow3" type="short">260</attr>
                        <attr key="Exposure" type="int">7000</attr>
                        <attr key="FrameDelay" type="int">1000</attr>
                        <attr key="GlobalShutter" type="bool">true</attr>
                        <attr key="NullSettle" type="short">6</attr>
                        <attr key="RampReset" type="short">20</attr>
                        <attr key="RampShortReset" type="bool">false</attr>
                        <attr key="ResetRead" type="bool">true</attr>
                        <attr key="ResetSettle" type="short">20</attr>
                        <attr key="RowSettle" type="short">20</attr>
                        <attr key="Run" type="bool">true</attr>
                        <attr key="SampleEnable" type="bool">true</attr>
                        <attr key="SampleSettle" type="short">60</attr>
                        <attr key="StartColumn0" type="short">0</attr>
                        <attr key="StartColumn1" type="short">346</attr>
                        <attr key="StartColumn2" type="short">346</attr>
                        <attr key="StartColumn3" type="short">346</attr>
                        <attr key="StartRow0" type="short">0</attr>
                        <attr key="StartRow1" type="short">260</attr>
                        <attr key="StartRow2" type="short">260</attr>
                        <attr key="StartRow3" type="short">260</attr>
                        <attr key="TakeSnapShot" type="bool">false</attr>
                        <attr key="UseInternalADC" type="bool">true</attr>
                        <attr key="WaitOnTransferStall" type="bool">true</attr>
                    </node>
                    <node name="bias" path="/1/1-DAVISFX3/DAVIS346B/bias/">
                        <node name="AEPdBn" path="/1/1-DAVISFX3/DAVIS346B/bias/AEPdBn/">
                            <attr key="coarseValue" type="byte">6</attr>
                            <attr key="currentLevel" type="string">Normal</attr>
                            <attr key="enabled" type="bool">true</attr>
                            <attr key="fineValue" type="short">91</attr>
                            <attr key="sex" type="string">N</attr>
                            <attr key="type" type="string">Normal</attr>
                        </node>
                        <node name="AEPuXBp" path="/1/1-DAVISFX3/DAVIS346B/bias/AEPuXBp/">
                            <attr key="coarseValue" type="byte">4</attr>
                            <attr key="currentLevel" type="string">Normal</attr>
[... 446 lines cut ...]
                <attr key="transferBufferSize" type="int">128</attr>
                <attr key="validOnly" type="bool">false</attr>
            </node>
        </node>
        <node name="2" path="/2/">
            <node name="1-AmbergTrolley" path="/2/1-AmbergTrolley/">
                <attr key="doSaveTxt" type="bool">true</attr>
                <attr key="doSavetxt" type="bool">true</attr>
                <attr key="runAtStartup" type="bool">true</attr>
[... cut ...]
            <attr key="logFile" type="string">/Users/federicocorradi/inilabs/caer_git/caer.log</attr>
            <attr key="logLevel" type="byte">5</attr>
        </node>
        <node name="server" path="/server/">
            <attr key="backlogSize" type="short">5</attr>
            <attr key="concurrentConnections" type="short">5</attr>
            <attr key="ipAddress" type="string">127.0.0.1</attr>
            <attr key="portNumber" type="int">4040</attr>
        </node>
    </node>
</sshs>

llongi, 2016-06-11:
Config looks okay. But the error changed now, it's about an invalid header. Not sure what you got there, take a look with a hex-editor. I'll try to reproduce a few things here. Eventually send me the file or link it via GDrive or DropBox.

federicohyo, 2016-06-11:
Hi Luca, this is not urgent. However the recordings are under GDrive->RailScanPrj->Data->LaserClassI_10062106->TakeI

this is reproducible under many machines... maybe something wrong at the header.

thank you for your help

llongi, 2016-06-12:
Cannot reproduce. Just took a new recording now and played it back with no issues.
I'll try to find the time to look at the file in question more closely. Only thing I can think of given the message is the file being corrupted somehow, maybe something doing some newline conversion between the Unix and Windows formats...

federicohyo, 2016-06-13:
Indeed, the input module is working fine (I just took a small recording and played it again). 
The recordings in gDrive are somehow corrupted, maybe when uploading them from my MacOsX.
Thanks Luca and sorry for the troubles. 

I am closing the issue.

closed 2016-06-13 by federicohyo