inivation/caer #15: data exchange between cAER with a C or C++-application?

Thanh-Binh, 2016-08-29:
Hi,
in cAER we can collect all events into images and save images somehows.
For live-demonstration with an standalone application, we need to get data from camera at the application framework.
 
My questions: 
1. how can we share or data exchange in the best way?
2. is there any interface available for data exchange, e.g. over shared memory?
Thanks

llongi, 2016-09-20:
To share data on the local machine, the easiest way would probably be to use the Unix Socket output module, which operates like a file output module, but sends the stream of data over a local Unix Socket.
See modules/misc/out/unix_socket.h for the interface, it's used like any other output module.

Thanh-Binh, 2016-09-21:
@llongi 
we used unix_socket_server.c for outputing, but by running the call "bind()" at line 54 failed.
if (bind(serverSockFd, (struct sockaddr *) &unixSocketAddr, sizeof(struct sockaddr_un)) < 0)

We really do not know why?
Is it any bug here?

llongi, 2016-09-26:
Yes, I agree that to exchange data locally between cAER and some other process, a local Unix socket is the easiest and most performant way. There is a discussion about it here:
https://sourceforge.net/p/jaer/discussion/631958/thread/cbdfb9af/
Closing this ticket then.

closed 2016-09-26 by llongi

Thanh-Binh, 2016-09-26:
thanks.
my student try to use Unix socket for sending a struct data like:
struct mydata
{
    int x[43200];
    int y[43200];
    int  dataLength;
}
and use send() and recv().
But it send and receive correctly if for a small data (e.g. using 432 instead of 43200).
Do you know this problem? Is there any data volume limit here?
Best thanks.

llongi, 2016-09-26:
For AF_UNIX SOCK_STREAM sockets, size shouldn't matter. But it you're receiving data from cAER's output module, you will have to parse it according to the AEDAT 3.1 format with a network header in front. Look at utils/unixststat/ for a simple example.

breznak, 2016-10-05:
Sorry for reopening the issue, but I have a similar problem... 

- So Sockets are suggested method for IPC in cAER (compared to SHM)? 
- By chance, is there an example implementation of a client reading the data from socket in C++? 
- regarding your answer for requirement of parsing the AEDAT 3.1 format, I assume it's done in https://github.com/inilabs/caer/blob/master/utils/unixststat/unixststat.c#L102 Would you consider using `ProtocolBuffers` or similar to make handling the AEDAT format easier in other languages? 

Thanks in advance,

llongi, 2016-10-06:
Well, between different processes, Unix Sockets (or named Pipes on Windows) are the supported way of doing it. SHM could be more performant, but we don't provide examples or support for it.
The utils/unixststat/ utility should give a useable example.
unixststat.c does some rudimentary parsing, the full format is described in http://inilabs.com/support/software/fileformat/ and currently I first want to finish up all the current code and modules, before thinking about other file formats or description systems. We're not really serializing the data, as we send the raw memory in most cases, around or to file, so not sure how much GProtoBufs would help.