inivation/caer #16: APS Video for external application?

Thanh-Binh, 2016-09-20:
I have some questions:
how can I ask cAER 
1. if a APS video data at a time instance is available for capturing into the application?
2. if it is available, I can get those data.?

do you have any mechanismus for avoiding read from and write to the same data (e.g. video data for APS and ) at the same time?

llongi, 2016-09-20:
I'm not exactly sure what you mean here. Do you mean writing an AVI made up of the APS frames? That could be done using [OpenCV's VideoWriter](http://docs.opencv.org/3.0-alpha/modules/videoio/doc/reading_and_writing_video.html#videowriter) for example. Or any other C library that can write videos out of raw frames, which is what APS frames are in the end.

Thanh-Binh, 2016-09-21:
No, I do not mean VideoWriter.
I mean APS video should be sent anyhow into application.
I think, using unix socket for sending video data to application maybe the best way.

llongi, 2016-09-26:
Yes, I agree that to exchange data locally between cAER and some other process, a local Unix socket is the easiest and most performant way. There is a discussion about it here:
https://sourceforge.net/p/jaer/discussion/631958/thread/cbdfb9af/
Closing this ticket then.

closed 2016-09-26 by llongi