inivation/caer #17: Bug: CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_PACKET_SIZE undeclared

breznak, 2016-09-20:
I recently get this error.
Env:
latest `libcaer`,
Ubuntu linux, compiling for `DAVISFX2`, ...

```
/mnt/store/devel/vision/dvs/caer/modules/ini/davis_common.c: In function ‘systemConfigSend’:
/mnt/store/devel/vision/dvs/caer/modules/ini/davis_common.c:2298:2: error: ‘CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_PACKET_SIZE’ undeclared (first use in this function)
  CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_PACKET_SIZE, U32T(sshsNodeGetInt(node, "PacketContainerMaxPacketSize")));
```

llongi, 2016-09-20:
Those are part of the newest libcaer. As far as I can tell, it's picking up an old install of libcaer.
Make sure you also 'make install' it, and that there is only one install on your system (like /usr VS /usr/local, sometimes you get two if you forget the INSTALL_PREFIX when doing libcaer's cmake as per its README).

breznak, 2016-09-20:
Indeed @llongi , thanks for the advice! I have libcaer installed in `/usr/` and `/usr/local/` 
Is there a better way to remove it than `find /usr -name "libcaer*"` and rm?

breznak, 2016-09-20:
Solved by removing the duplicit (outdated) library. Thanks!

closed 2016-09-20 by breznak