inivation/caer #18: Bug: undefined reference to caerEventPacketContainerGetHighestEventTimestamp, caerSpecialEventPacketFindEventByType

breznak, 2016-09-20:
Building master fails on 
```
[  4%] Linking C executable caer-bin
CMakeFiles/caer-bin.dir/modules/ini/davis_common.c.o: In function `caerInputDAVISRun':
/mnt/store/devel/vision/dvs/caer/modules/ini/davis_common.c:(.text+0x75eb): undefined reference to `caerEventPacketContainerGetHighestEventTimestamp'
/mnt/store/devel/vision/dvs/caer/modules/ini/davis_common.c:(.text+0x7626): undefined reference to `caerSpecialEventPacketFindEventByType'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

This is after I "workarounded" #17 (by defining it on it's first usage). 

In a more detail, I get a warning about invalid C99 "implicit function declaration":
```
[ 48%] Building C object CMakeFiles/caer-bin.dir/modules/ini/davis_common.c.o
/mnt/store/devel/vision/dvs/caer/modules/ini/davis_common.c:302:4: warning: implicit declaration of function 'caerEventPacketContainerGetHighestEventTimestamp' is invalid in
      C99 [-Wimplicit-function-declaration]
                        caerEventPacketContainerGetHighestEventTimestamp(*container));
                        ^
/mnt/store/devel/vision/dvs/caer/modules/ini/davis_common.c:308:8: warning: implicit declaration of function 'caerSpecialEventPacketFindEventByType' is invalid in C99
      [-Wimplicit-function-declaration]
                        && (caerSpecialEventPacketFindEventByType((caerSpecialEventPacket) special, TIMESTAMP_RESET) != NULL)) {
```

llongi, 2016-09-20:
Those are part of the newest libcaer. As far as I can tell, it's picking up an old install of libcaer.
Make sure you also 'make install' it, and that there is only one install on your system (like /usr VS /usr/local, sometimes you get two if you forget the INSTALL_PREFIX when doing libcaer's cmake as per its README).

breznak, 2016-09-20:
Solved by advice above, thanks!

closed 2016-09-20 by breznak