inivation/caer #19: File input: add support for AEDAT v2.0

breznak, 2016-09-22:
```
ERROR: 10-FileInput: Reading AEDAT 2.0 data not yet supported.
ERROR: 10-FileInput: Failed to parse event data.
```

Do you know an estimate when the functionality could be available? 

How can I workaround it for now? And record with Davis240 in the "old" format? (by using `-DJAER_COMPAT_FORMAT=1` ?) 

Thank you,

llongi, 2016-09-26:
With the current input modules, there is no support/workaround to get the old AEDAT 2.X format as output. But the AEDAT 3.x output should be able to be opened in jAER (that's all the JAER_COMPAT_FORMAT flag was good for). If you can't open cAER AEDAT 3.X files with jAER, please open a thread on the jAER SF.net forum, so I can refer the person in charge of that piece of code to take a look.
Regarding INPUT support for jAER AEDAT 2.X, it's on my TODO list. Depending on other projects and time availability, should be done by mid-end October.

breznak, 2016-09-26:
Thank you for the clarification. This has nothing to do with jAER, in cAER i was unable to open and replay the files using the recent "file input". Do I understand the situation that cAER will support only the new (3.x) AEDAT format? 

I'll retest with re-recording a file directly from caer. 

Sidenote, are you considering a convertor tool to convert files from the older format(s) (if that's possible)?

breznak, 2016-09-26:
> are you considering a convertor tool to convert files from the older format(s) (if that's possible)? 

ok, I think I got the answers: 
- AEDAT 1.0 (.dat, binary header) - unsupported
- 2.0 : TODO
- 3.0: supported in cAER

- conversion is possible in jAER

llongi, 2016-09-26:
Yes, that's a correct understanding of things on the cAER input side. On the output side, it will always be AEDAT 3.X. Conversion of 1,2,3 to 2.X is possible in jAER. Conversion of 2,3 to 3.X will be possible in cAER.

breznak, 2016-09-26:
Maybe a small suggestion, it might be nice to have a (few) demo files of each version available on the official site (I haven't looked recently though). 

// please feel free to close, or keep this for tracking AEDAT2.0 in cAER...

maxwellflitton, 2017-01-20:
Is there Python support for reading AEDAT files?

tobidelbruck, 2017-01-21:
Take a look in the jaer folder ~jaer-trunk\scripts\python\jAER_utils

https://sourceforge.net/p/jaer/code/HEAD/tree/scripts/python/

specifically 
https://sourceforge.net/p/jaer/code/HEAD/tree/scripts/python/jAER_utils/

There are basic scripts there for reading and writing AER-DAT-2.0 files, 
but they may need some debugging and updating.  You are very welcome to 
join sourceforge and ask us for commit rights to commit improvements to 
these scripts.

Best, Tobi

maxwellflitton, 2017-02-10:
I have managed to get the logging data script working on python 3. I am working through the net_raw, it seems to be working fine until I get to the line:

data = s.recvfrom(buf_size)

It just stops... it doesn't crash, it just doesn't do anything. Is this line supposed to take a long time to execute?

federicohyo, 2017-02-10:
Hi Maxwell,

it might be that buf_size is very large, how is it calculated? which camera
are you using? If buf_size is large, then the functions waits to receive
exactly buf_size bytes.

Make sure that the buffer size is calculated as described in the file
format: https://inilabs.com/support/software/fileformat/#h.w7vjqzw55d5b

let me know if this help.

with best regards,
Federico

maxwellflitton, 2017-02-10:
I'm using a DVS 128. In the logging function, you guys give a buffer size of 1024. However, when unpacking it you get the error: "struct.error: unpack requires a bytes object of length 4". On the net_raw function, you guys give a buffer size of 63000. This is where it simply takes forever to execute the command of:

data = s.recvfrom(buf_size)

federicohyo, 2017-02-10:
Hi maxwell,

Are you only running the python script or also caer? It looks like you
don't have any data. Did you compile caer with network output?

Best,
Federico

maxwellflitton, 2017-02-10:
No, I didn't know I had to use caer. Is there a readme because this adds another level to the solution. Do I have to run a certain caer script and then run the python script I've been talking about?

federicohyo, 2017-02-10:
Hi Mawell,

I can produce a small example and a readme for you, next week (ie. Monday).

However, what are you trying to do? If you wish to analyse data from the
sensor maybe you can just run jAER, make recordings and then load them in
python.

Another option is to directly use cAER and make recordings with it. Which
system are you using and what is the general goal of your project? I may be
able to help you better.

In addition have you had a look at our software page?
https://inilabs.com/support/software/

with best regards,
 Federico

maxwellflitton, 2017-02-10:
I am using windows. I am doing an MSc in physics and engineering in medicine at UCL. I want to load live data into python so we can start detecting points of interest and developing 3D mapping in real time. We think that this sensor could revolutionize surgical robotics. The hurdle that I need help with is parsing the data from the sensor into the python program whilst the sensor is running. Once this is done we can start implementing our algorithms and tuning the sensor for a surgical environment.

tobidelbruck, 2017-02-11:
You should be able to read live data from a running AEViewer into 
python, using a UDP port connection.

In the running AEViewer, you can select the menu item 
"File/Remote/Enable unicast datagram (UDP) output....".

In the resulting dialog you can select options for the output.

On the receiving side, you must parse the received data. For a matlab 
example, see the script readaeudpstream.m in the folder <jaer 
trunk>\scripts\matlab. In addition, the received raw data must be 
decoded to events, APS samples, frames, etc. This parsing can get rather 
complex for DAVIS frame data. See the matlab examples in matlab/davis 
folder as a guide.

There are some python examples as well but I'm not familiar with them. 
Take a look at scripts\python\cAER_utils. Note these assume you have a 
running cAER that is sending camera data to a UDP port. The format of 
data sent from caaer follows the AEDAT-3.1 format. See 
https://inilabs.com/support/software/fileformat/ for authoritative 
details on all formats.

maxwellflitton, 2017-02-23:
Sorry for the late reply. I had a paper due in. I've selected: 

"File/Remote/Enable unicast datagram (UDP) output...."

I then ran my main.py file and got some output but still the same error again at the end. I have attached my main.py file and the output as a word file. The buff size is the dafault buff size that was there when I selected: 

"File/Remote/Enable unicast datagram (UDP) output...."

so I put this in my main.py script. There's a lot of print functions in the script as I just want to see what's coming out. The first part is logging the data that doesn't seem to throw up any errors. The second part is parsing the data into the python module.
[main.py.zip](https://github.com/inilabs/caer/files/795206/main.py.zip)
[output.docx.zip](https://github.com/inilabs/caer/files/795208/output.docx.zip)

maxwellflitton, 2017-02-23:
I appreciate that my coding skillset is not very strong. Although I have coded web apps in a Django framework my undergrad was physics. When we get this working would you tolerate me writing idiots guide for other people like me? If we manage to succeed in showing that it has a use in surgical applications I want others with similar backgrounds to me in the medical physics and bioengineering academic departments to hit the ground running with using it in python.

federicohyo, 2017-02-23:
Hi,

Maxwell I believe this discussion is more for the jAER forum. It is definitely not an issue for the current topic.
Please post your comment / request for help in here: https://sourceforge.net/p/jaer/discussion/631958/

thank you