inivation/caer #23: add option DRYRUN=n to limit now events processed

breznak, 2016-09-22:
as `-DDRYRUN=1000` to run only for given number of events(/seconds/...?)

The motivation is to allow for testing and running the program for only a limited number of events.

llongi, 2016-09-26:
This is an interesting idea. I'm thinking more of two parameters in the mainloop configuration (so at the top-node of caer-config XML), something like "shutdownAfterEvents" (sum of all events) and "shutdownAfterTTime" (time passed between first and last event), disabled by default (0), where if you set them, as soon as one of the two limits is hit, it shuts cAER down. What do you think? I think this is more flexible than a compile-time option and more aligned with how we usually treat time/size limits in cAER.

breznak, 2016-09-26:
> something like "shutdownAfterEvents" (sum of all events) and "shutdownAfterTTime" (time passed between first and last event), disabled by default (0) 

Yes, that will be a good solution :+1: