inivation/caer #24: Key shortcut to exit the program (correctly)

breznak, 2016-09-22:
Eg. Esc , or ctrl^c should quit the program successfully. Now, when I run with the HW camera, I can only ctrl^z to suspend it thread and then `kill -9 %1` to kill it. But then, the prog segfaults and caer-config.xml is not written, etc.. 

Similar problem of limiting number of events/time a live program runs for is in #23

llongi, 2016-09-26:
That's strange, CTRL+C is explicitly supported, we do install the appropriate signal handlers exactly to support quitting by CTRL+C, and I'm using it dozen of times daily, on multiple systems, and it always works. How are you compiling cAER (all the options)? Any extra modules that maybe hang?

breznak, 2016-09-26:
> ... compiling cAER (all the options)? Any extra modules that maybe hang? 

May be internal module then. Will test with the upstream modules only. I suspect the Visualizer might be the problem (it also spawns a separate process)

breznak, 2016-09-29:
Yes, confirmed it's caused by the private module. Thank you and closing. 

Btw, @llongi , do you have a clue why the official modules react fine to ctrl^C and mine segfaults? (I know, hard to tell without looking at the code)

closed 2016-09-29 by breznak

llongi, 2016-10-03:
CTRL+C shuts everything doen, so it calls all the Exit() functions. Maybe your module's has some bug. Double free? Freeing wrong resources? Something like that.