inivation/caer #25: Controlling packet size from config

breznak, 2016-09-23:
Related to the issue https://github.com/inilabs/caer/issues/8 

First of all, could you please summarize how is EventPacket size limited? For 
"nothing-happens" sceneses I get packets with 10s of events only, when I then perform a sudden movement, I could end up to with 10.000s events in a single packet! 

The size limit is enforced by `CAER_EVENT_PACKET_HEADER_SIZE`, right? Is there also a time limit in ms? 

Can these two variables be made user-configurable in `caer-config.xml`?

breznak, 2016-09-23:
Are there any known bad implications of setting the packet size ridiculously small, eg 20?

llongi, 2016-09-26:
CAER_EVENT_PACKET_HEADER_SIZE is a constant set to 28 (bytes). It has nothing to do with the number of events in a packet, it just tells how big the common packet header is.
The "packeting system" works in this way: events are accumulated by type in a packet, and that packet is part of a packet container, by an input module.
The packet container is then sent out for processing when either the configured time limit or the size limit are hit.
The time limit is always active, in microseconds, and basically tells you the time-span an event packet covers. This enables regular, constant delivery of packets, that cover a period of time.
The size limit is an addon to prevent packets to grow to immense sizes (like if the time limit is high and there is lots of activity). As soon as a packet hits the number of events in the size limit, it is sent out. The regular time limit is not reset in this case. This size limit can be disabled by setting it to 0.
The DVS/DAVIS input modules call these two configuration variables "PacketContainerMaxInterval" and "PacketContainerMaxPacketSize".
The File/Network input modules call them "timeSlice" and "sizeSlice".
Too small packet sizes or intervals simply mean more packets, which may negatively affect performance. It's usually a good idea to set the size to something around 4-8K, and the time to a good value based on the application you're building, so if you need ms-reaction-time, you probably want to set it to 1000µs, so that you do get new data every ms. If on the other hand you're looking at a static scene and just want to detect that something is passing by once every while, a high number like 100ms might also be perfectly fine.
I hope this helps in understanding how we bundle up events.

breznak, 2016-09-29:
This is a great explanation, than you very much Luca!! 
Maybe it would be worth adding to some readme? 

> The DVS/DAVIS input modules call these two configuration variables "PacketContainerMaxInterval" and "PacketContainerMaxPacketSize".
The File/Network input modules call them "timeSlice" and "sizeSlice".

What do you think about making those 2 the same?

llongi, 2016-10-03:
Yes, that's a good idea. DVS128, DAVIS and Input modules now have the same name (in libevent_windows branch, so that will propagate to master when I merge back).
"PacketContainerInterval" and "PacketContainerMaxPacketSize" are now used everywhere, and for consistency I changed "timeDelay" in the Input module to also be "PacketContainerDelay".
I've also updated libcaer's packetContainer.h documentation to include the above information, this way it will automatically be part of the libcaer manual.

closed 2016-10-03 by llongi