inivation/caer #26: building error

Thanh-Binh, 2016-10-05:
I have just download the current version, but have the following error:
/home/binh/Neuromorphics/caer/modules/ini/davis_common.c: In function ‘systemConfigSend’:
/home/binh/Neuromorphics/caer/modules/ini/davis_common.c:2298:2: error: ‘CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_PACKET_SIZE’ undeclared (first use in this function)
  CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_PACKET_SIZE, U32T(sshsNodeGetInt(node, "PacketContainerMaxPacketSize")));

where and how big is CAER_HOST_CONFIG_PACKETS_MAX_CONTAINER_PACKET_SIZE defined?
thanks

llongi, 2016-10-05:
You also need to update libcaer to the latest version. And make sure you only have one active install of libcaer on your system (sometimes it happens that people forget to pass the -DCMAKE_INSTALL_PREFIX=/usr and end up with an install in /usr and one in /usr/local).

closed 2016-10-05 by llongi

Thanh-Binh, 2016-10-05:
I have upgrade libcaer and install it firstly.
Pls make sure that both libcaer and caer are at the same folder.
after that I build caer using
cmake  -DDAVISFX3=1   -DENABLE_BAFILTER=1 -DENABLE_STATISTICS=1 -DENABLE_VISUALIZER=1  . && make -j8 

Do I have to configure somehows to let compiler know where is include file of libcaer?

llongi, 2016-10-05:
No, it should find it on its own if it's correctly installed. You may need to clear the CMake cache:
rm -Rf CMakeFiles/ CMakeCache.txt