inivation/caer #27: UDP network output crashes

breznak, 2016-10-06:
I'm using the default config and main.c
File recorded on FX2 and replayed through the FILE_INPUT

* TCP: 
 - first start the server `./caer-bin` 
 - then "client" ` ./utils/tcpststat/tcpststat`
 - runs OK and outputs like 
```
type = 3, source = 10, size = 36, tsOffset = 4, tsOverflow = 0, capacity = 10, number = 10, valid = 10.
Time difference in packet: 9011 (first = 10947413, last = 10956424).

type = 0, source = 10, size = 8, tsOffset = 4, tsOverflow = 0, capacity = 2, number = 2, valid = 2.
Time difference in packet: 1000 (first = 10954479, last = 10955479).
```

* UDP
- I must first run the client (?) ` ./utils/udpststat/udpststat ` then the server
- sometimes, the server spawns warnings: 
```
Total events/second:       1049 - Valid events/second:       10492016-10-06 16:52:39 (TZ+0200): ERROR: 8-NetTCPServerOutput: Detected timestamp going back, expected at least 68086288 but got 1126628. Ignoring packet of type 1 from source 10, with 105 events!
2016-10-06 16:52:39 (TZ+0200): ERROR: 8-NetTCPServerOutput: Detected timestamp going back, expected at least 68086288 but got 1135575. Ignoring packet of type 2 from source 10, with 1 events!
2016-10-06 16:52:39 (TZ+0200): ERROR: 8-NetTCPServerOutput: Detected timestamp going back, expected at least 68086288 but got 1127418. Ignoring packet of type 3 from source 10, with 10 events!
```
- and the client crashes with 
```
Result of recv() call: 9572
Magic number: 2105305046418351704
Sequence number: 0
Version number: 1
Format number: 0
Source number: 10
type = 11, source = 372, size = 6684, tsOffset = 13369875, tsOverflow = 6684, capacity = 13238451, number = 6684, valid = 24248335.
2016-10-06 16:47:55 (TZ+0200): CRITICAL: Generic Event: Called caerGenericEventGetEvent() with invalid event offset 24248334, while maximum allowed value is 13238450. Negative values are not allowed!
Segmentation fault (core dumped)
```

Any idea about this?

llongi, 2016-10-06:
Yes, UDP output is currently broken. I'm rewriting the networking part for Windows support in the libevent_windows branch, so I'm also making fixing and testing all the outputs as parts of that work. When I'll merge the branch back it will get fixed, leaving this open for now.

breznak, 2016-10-06:
Btw, are you programming the library yourself, or outsourcing the task to an existing lib? Just came across https://github.com/dermesser/libsocket and it looks fine

llongi, 2016-10-06:
No, I'm using [libuv](http://libuv.org/). It's the only one I found that supports Windows, Linux and OS X, is a C library, is well supported, and also abstracts UDP and UnixSockets/Pipes. I have to change a few things around to accomodate for the asynchronous I/O nature of it, but nothing too terrible and in the end that might even help performance. The libsocket library you link doesn't seem to support Windows. Maybe Cygwin, but I'm targeting MinGW, so Windows+GCC but without the big Cygwin emulation layer.

Thanh-Binh, 2016-10-16:
@llongi when do you plan to make UDP available? I understand your code that UDP uses for send/receive APS frame over network?Am I right?

llongi, 2016-10-18:
The rewritten output code is now in, using libuv. This also should work on Windows. TCP, and Unix Sockets seem to be working fine. UDP needs some more work in the udpststat.c to show a good example, since decoding it is much more complex than TCP/Sockets, as it's not a stream you have to keep track of the sequence number and rebuild packets out of UDP messages.