inivation/caer #31: problems by tcpststate.c

Thanh-Binh, 2016-10-18:
I found the following problem by TCP-Data:
1. the x and y coordinate data are exchanged, that means
caerPolarityEvent currEvent = caerPolarityEventPacketGetEvent(header, i);
int y   = caerPolarityEventGetX(currEvent);
int x   = caerPolarityEventGetY(currEvent);

Normally, they should look like
int x   = caerPolarityEventGetX(currEvent);
int y   = caerPolarityEventGetY(currEvent);

2. sometime, a valid event has x or y which are bigger than image size. In my test w = 240, h= 180 for DAVIS240.

3.  Both server and client have sometime ERROR
Total events/second:        273 - Valid events/second:        2572016-10-18 16:34:16 (TZ+0200): ERROR: 8-NetTCPServerOutput: Detected timestamp going back, expected at least 68086288 but got 3276511. Ignoring packet of type 1 from source 10, with 489 events!
2016-10-18 16:34:16 (TZ+0200): ERROR: 8-NetTCPServerOutput: Detected timestamp going back, expected at least 68086288 but got 3277036. Ignoring packet of type 3 from source 10, with 10 events!
2016-10-18 16:34:16 (TZ+0200): ERROR: 9-NetUDPOutput: Detected timestamp going back, expected at least 68086288 but got 3276511. Ignoring packet of type 1 from source 10, with 489 events!

Has anyone a solution idea?

llongi, 2016-10-18:
To 1 and 2, that should not happen. Can you try again with the libuv-based code I just merged into master? I think the previous code had a possible bug where it sent out a packet in the middle, resulting in corrupt memory. Also when you get the "timestamp going back" error, where does the data come from? A device or a recorded file?

breznak, 2016-10-18:
> Can you try again with the libuv-based code I just merged into master? I think the previous code had a possible bug where it sent out a packet in the middle, resulting in corrupt memory.  

Interesting, I had a similar problem when writing a C++ socket client (using ZeroMQ), I got a packet of 20B, then data (82xx B), then 5B (empty?) packet, 82xx B, ... Will revisit the code again..

Thanh-Binh, 2016-10-18:
@llongi: I believe it is a bug in the function caerPolarityEventGetX,Y. I will testtomorow.
Do you have a code for testing client receiving aps frame?

Thanh-Binh, 2016-10-19:
@llongi I have just updated libcaer, and caer + libuv as you mentioned before.
But we have the same problems.
In the client side, there is an error like:
tcpststat: malloc.c:2395: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

Do you know where does it error come from?

llongi, 2016-10-19:
That's clearly some form of memory corruption. Probably a double or invalid free(). But I can't reproduce it by using cAER and tcpststat.c, no errors and no memory leaks. Did you modify the tcpststat.c code in any way? If yes, please post the exact code.

Thanh-Binh, 2016-10-19:
it is solved. It is a mistake from my side. Sorry
Best thanks.

closed 2016-10-19 by Thanh-Binh