inivation/caer #35: segfault on libuv when networkoutput module is enable

federicohyo, 2016-10-28:
federico@zfractal:/media/federico/space/inilabs/caer$ gdb caer-bin 
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.04) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from caer-bin...(no debugging symbols found)...done.
(gdb) r
Starting program: /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/caer-bin 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
2016-10-28 11:11:10 (TZ+0200): NOTICE: Logger: Initialization successful with log-level 5.
[New Thread 0x7fffe54e5700 (LWP 6472)]
2016-10-28 11:11:10 (TZ+0200): ERROR: Config Server: uv_listen failed, error -98 (EADDRINUSE).
[New Thread 0x7fffe4ce4700 (LWP 6474)]
[Thread 0x7fffe54e5700 (LWP 6472) exited]
[New Thread 0x7fffdffff700 (LWP 6475)]
[New Thread 0x7fffdf7fe700 (LWP 6476)]
2016-10-28 11:11:10 (TZ+0200): ERROR: 8-NetTCPServerOutput: uv_listen failed, error -98 (EADDRINUSE).

Thread 3 "Mainloop-1" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe4ce4700 (LWP 6474)]
0x00007ffff7574a00 in ?? () from /usr/lib/x86_64-linux-gnu/libuv.so.1

llongi, 2016-10-28:
Installed version of libuv? Does this always happen or only when you restart cAER quickly and it somehow didn't clean up the previous connections (I see several EADDRINUSE errors)?

federicohyo, 2016-12-17:
see also this ... https://sourceforge.net/p/jaer/discussion/631958/thread/18561b05/?limit=25#6a10

llongi, 2017-02-17:
Fixed in commit 5d25ca8bb23b17a9f840e0f41b3997b02cd5724f.

closed 2017-02-17 by llongi