inivation/caer #36: Make caer/ext/sshs/sshs_node.c error.

mrunix, 2016-12-05:
caer/ext/sshs/sshs_node.c:45:50: error: unknown type name ‘mxml_node_t’
 static const char *sshsNodeXMLWhitespaceCallback(mxml_node_t *node, int where);

I grep MXML libirary, I can not find the mxml_node_t type define.

federicohyo, 2016-12-05:
Hi, 

did you install libmxml from here http://www.msweet.org/projects.php?Z3 ?

you need: "mini-xml (mxml) >= 2.7 (development package); then run ./configure; then run “make”; then run “sudo make install”)"

I am sorry but I can't reproduce that error, I just did a fresh install on a new ubuntu 16.04 and it works fine. In which system are you currently trying to install caer?

mrunix, 2016-12-06:
OK.  My errors. The version of mxml library wrong. I fixed it.

closed 2016-12-06 by federicohyo