inivation/caer #37: Make caer main.c errors

mrunix, 2016-12-06:
caer/main.c:180:114: error: ‘frame’ undeclared (first use in this function)
 zerRendererFrameEvents, visualizerEventHandler, (caerEventPacketHeader) frame);
caer/main.c:180:114: note: each undeclared identifier is reported only once for each function it appears in
/works/projects/PositionDVS/app/caer/main.c:181:112: error: ‘imu’ undeclared (first use in this function)
 alizerRendererIMU6Events, visualizerEventHandler, (caerEventPacketHeader) imu);

CMAKE:
cmake  -DDVS128=1 -DENABLE_VISUALIZER=1 -DENABLE_STATISTICS=1

mrunix, 2016-12-06:
OK. The correct cmake is:
cmake -DDVS128=1 -DENABLE_VISUALIZER=1 -DENABLE_STATISTICS=1 -DENABLE_FILE_INPUT=1

federicohyo, 2016-12-06:
Not really, in the main.c example you can use either input from the camera or input from the file. 
You should therefore enable only one input -DDVS128=1 or -DENABLE_FILE_INPUT=1  and not both at the same time.

llongi, 2016-12-06:
Yes, this was a bug with -DDVS128=1, it's been fixed in the latest commit. You don't need -DENABLE_FILE_INPUT=1 if you want to use a DVS128 device.

closed 2016-12-06 by llongi