inivation/caer #38: Where are caerSpikeEventGetY and caerSpikeEventGetX?

mrunix, 2016-12-06:
[ 16%] Linking C executable caer-bin
[ 30%] Built target caer-ctl
[ 36%] Built target unixststat
CMakeFiles/caer-bin.dir/modules/visualizer/visualizer_renderers.c.o: In function `caerVisualizerRendererSpikeEventsFrequency':
**visualizer_renderers.c:(.text+0xa18): undefined reference to `caerSpikeEventGetY'
visualizer_renderers.c:(.text+0xa25): undefined reference to `caerSpikeEventGetX'**
CMakeFiles/caer-bin.dir/modules/visualizer/visualizer_renderers.c.o: In function `caerVisualizerRendererSpikeEvents':
visualizer_renderers.c:(.text+0xb51): undefined reference to `caerSpikeEventGetY'
visualizer_renderers.c:(.text+0xb6b): undefined reference to `caerSpikeEventGetX'
collect2: error: ld returned 1 exit status
CMakeFiles/caer-bin.dir/build.make:570: recipe for target 'caer-bin' failed
make[5]: *** [caer-bin] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/caer-bin.dir/all' failed
make[4]: *** [CMakeFiles/caer-bin.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make[3]: *** [all] Error 2
app/CMakeFiles/caer.dir/build.make:110: recipe for target 'app/caer-prefix/src/caer-stamp/caer-build' failed
make[2]: *** [app/caer-prefix/src/caer-stamp/caer-build] Error 2
CMakeFiles/Makefile2:93: recipe for target 'app/CMakeFiles/caer.dir/all' failed
make[1]: *** [app/CMakeFiles/caer.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

federicohyo, 2016-12-06:
Hi,

these functions have been recently added in libcaer.
Did you update libcaer from git?

mrunix, 2016-12-06:
I try it.

mrunix, 2016-12-06:
The latest codes is OK.

On Linux, if need display video, how make caer? And run which application?

closed 2016-12-06 by federicohyo

federicohyo, 2016-12-06:
In caer you should be able to record an output aedat file by enabling the outputmodule.. something like

cmake -DDVS128=1 -DENABLE_VISUALIZER=1 -DENABLE_STATISTICS=1 -DENABLE_FILE_OUTPUT=1 .

then when you run caer you can start and stop the recordings using ./utils/caerctl/caer-ctl (or use the javaFX2 GUI) https://github.com/inilabs/caerctl-gui-javafx.

once you  are done with the recordings and you want to play it back. you can compile (or have it compiled already) a caer version with the input module instead of the DVS128.. something like:

cmake  -DENABLE_FILE_INPUT=1 -DENABLE_VISUALIZER=1 -DENABLE_STATISTICS=1 -DENABLE_FILE_OUTPUT=1 .

and set the options of the file to be played by using caer-ctl or the javaFX2 gui