inivation/caer #40: trying to make caer on Mac OS X 10.11.6

duguyue100, 2016-12-14:
Hi, I'm trying to make `caer` on my mac. After generating Makefile using this

```
cmake -DDVS128=1 -DDAVISFX2=1 -DDAVISFX3=1 -DENABLE_FILE_INPUT=1 -DENABLE_NETWORK_INPUT=1 -DENABLE_NETWORK_OUTPUT=1 -DENABLE_BAFILTER=1 -DENABLE_STATISTICS=1 -DENABLE_VISUALIZER=1 -DENABLE_IMAGEGENERATOR=1 -DENABLE_CAMERACALIBRATION=1 -DENABLE_POSEESTIMATION=1 .
```

It hits me with two error when compiling `main.c.o` and `all`, the log is:

```
/Users/dgyHome/Documents/workspace/caer/main.c:142:23: error: redefinition of 'frame'
        caerFrameEventPacket frame = NULL;

/Users/dgyHome/Documents/workspace/caer/main.c:143:22: error: redefinition of 'imu'
        caerIMU6EventPacket imu = NULL;
```

I had all necessary dependencies installed besides caffe. And from the cmake log, I would say they are all detected successfully.

Is it because there is a option conflicts? And if it is, please give some suggestion, thanks :)

federicohyo, 2016-12-14:
Hi Yuhang,

you are trying to enable all the inputs DAVISFX2, DVS128, DAVISFX3,
file_input and network input. But only one of these options can be enabled
at once, see README.
Also note that pose estimation requires opencv 3.0 with opencv_contrib.

try this:

cmake -DDVS128=0 -DDAVISFX2=1 -DDAVISFX3=0 -DENABLE_FILE_INPUT=0
-DENABLE_NETWORK_INPUT=0 -DENABLE_NETWORK_OUTPUT=0 -DENABLE_BAFILTER=1
-DENABLE_STATISTICS=1 -DENABLE_VISUALIZER=1 -DENABLE_IMAGEGENERATOR=0
-DENABLE_CAMERACALIBRATION=0 -DENABLE_POSEESTIMATION=0 .

if this goes well than you can enable cameracalibration and
poseestimation... of course remember to install opencv >= 3.0 with
opencv_contrib

best,
 Federico.

electronically yours,
 Federico

duguyue100, 2016-12-14:
Thanks for the reply.
I was trying to enable all the available cameras just now.
So I have to build new caer for every camera?

duguyue100, 2016-12-14:
Ok, now it works. Thanks.

closed 2016-12-14 by duguyue100

federicohyo, 2016-12-14:
not necessary, you can have 3 pre-compiled version...

duguyue100, 2016-12-14:
Yeah, that's what I meant. Like three different kinds of camera.
Thanks for your help!