inivation/caer #44: error after restart playing ... for offline ...

Thanh-Binh, 2017-02-02:
Hi,
I have this problem for a long time, and hope it should be fixed now.
I play a aerdat-file, recorded from DAVIS640 some months ago.
From the beginning until the end, it works well.
Because I will that cAER plays again, by setting caer-config.xml like:

<node name="10-FileInput" path="/1/10-FileInput/">
                <attr key="autoRestart" type="bool">true</attr>
                <attr key="bufferSize" type="int">65536</attr>
                <attr key="filePath" type="string">/home/binh/Neuromorphics/data/head_moving.aedat</attr>
                <attr key="keepPackets" type="bool">false</attr>
                <attr key="pause" type="bool">false</attr>
                <attr key="runAtStartup" type="bool">true</attr>
                <attr key="sizeSlice" type="int">8192</attr>
                <attr key="timeDelay" type="int">10000</attr>
                <attr key="timeSlice" type="int">10000</attr>
                <attr key="transferBufferSize" type="int">128</attr>
                <attr key="validOnly" type="bool">false</attr>
            </node>

so that it shoud automatically restart playing.
But I see many errors like:

2017-02-02 09:29:00 (TZ+0100): ERROR: 8-NetTCPServerOutput: Detected timestamp going back, expected at least 24799997 but got 13144972. Ignoring packet of type 0 from source 10, with 2 events!
2017-02-02 09:29:00 (TZ+0100): ERROR: 9-NetUDPOutput: Detected timestamp going back, expected at least 24799997 but got 13140438. Ignoring packet of type 3 from source 10, with 10 events!

Could you please inform me the way to avoid it?
thanks

federicohyo, 2017-02-02:
Hi Binh,

the error is related to the network output module (TCP and UDP). 
Since you are not using them (as you are playing back a file) you should disable them from the caer-config.xml. Set <attr key="runAtStartup" type="bool">false</attr>, so that they won't run at all.
We are still working on the network output modules and they might not be stable at the moment.

In addition the DAVIS640 requires special attention to bias settings which we are still investigating, please let us know if you have any issues with those. 

best regards,
 Federico

Thanh-Binh, 2017-02-02:
Hi,
do you mean that I have to set the attribute "autoRestart" to false?
In this case, we can not replay file multiple time...

federicohyo, 2017-02-02:
I mean to change the property of the Network output module, not the properties of the file input module. You have the network output module enable but you are not using it.

llongi, 2017-02-02:
This is still a bug. There is support for detecting and handling timestamp resets, but it only works right now for resets that are part of an event stream (from camera or from file). The special case that arises when the reset comes from the file being played twice (autoRestart flag) is not handled in any way. It should be.

Thanh-Binh, 2017-02-02:
@llongi: thanks, it is very nice if this bug is fixed

Thanh-Binh, 2017-03-20:
@llongi is this bug already fixed? Thanks

llongi, 2017-03-21:
Not yet fixed.

breznak, 2017-04-21:
@llongi  ...follow up from the duplicate issue:

> The right way to do it is to insert a Timestamp Reset event packet to delimit the stream before and after, exactly how it's handled for device-side resets or resets inside a stream. All clients need to handle that correctly anyway (the Input modules do so). This also triggers the proper resets inside modules.

...I'm going to look into it, thanks for the pointers