inivation/caer #51: cAER on Windows 10 - Config server : uv_listen failed (-4091)

federicohyo, 2017-03-12:
Windows 10 with mingw64. 

The configuration server does not start correctly, even if netstat does not shows any use of the 4040 tcp port. Is this a permission problems?

$ ./caer-bin.exe
2017-03-12 21:30:16: NOTICE: Logger: Initialization successful with log-level 5.
**2017-03-12 21:30:16: ERROR: Config Server: uv_listen failed, error -4091 (EADDRINUSE).**
Total events/second:         19 - Valid events/second:          5

netstat does not show any use of the port 4040.

inilabs@DESKTOP-BBC4NNK MINGW64 /home/caer
$ NETSTAT.EXE -f

Active Connections

  Proto  Local Address          Foreign Address        State
  TCP    192.168.0.127:59710    db5sch101110618:https  ESTABLISHED
  TCP    192.168.0.127:59735    a104-66-164-171:https  ESTABLISHED
  TCP    192.168.0.127:59872    lhr35s05-in-f67:https  ESTABLISHED
  TCP    192.168.0.127:59878    fra15s09-in-f14:https  ESTABLISHED
  TCP    192.168.0.127:59880    fra15s09-in-f45:https  ESTABLISHED
  TCP    192.168.0.127:59882    fra16s08-in-f202:https  ESTABLISHED
  TCP    192.168.0.127:59884    fra15s09-in-f14:https  CLOSE_WAIT
  TCP    192.168.0.127:59885    fra15s09-in-f35:https  ESTABLISHED
  TCP    192.168.0.127:59886    fra15s09-in-f35:https  ESTABLISHED
  TCP    192.168.0.127:59887    fra15s18-in-f3:https   ESTABLISHED
  TCP    192.168.0.127:59897    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59902    192.30.253.116:https   TIME_WAIT
  TCP    192.168.0.127:59904    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59905    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59906    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59907    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59916    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59917    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59929    ec2-54-210-94-233:https  ESTABLISHED
  TCP    192.168.0.127:59931    192.30.253.112:https   TIME_WAIT
  TCP    192.168.0.127:59933    ec2-54-210-94-233:https  TIME_WAIT
  TCP    192.168.0.127:59934    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59935    151.101.112.133:https  ESTABLISHED
  TCP    192.168.0.127:59936    192.30.253.125:https   ESTABLISHED
  TCP    192.168.0.127:59940    fra15s09-in-f14:https  SYN_SENT
  TCP    192.168.0.127:59941    fra15s09-in-f14:https  SYN_SENT

federicohyo, 2017-03-12:
It is related to the way in which caer-bin.exe was closed previously. Not a bug at all, caer-bin must be closed with control+c and not by closing the window by pressing the X on the top right corner.

closed 2017-03-12 by federicohyo

llongi, 2017-03-12:
Yes, if I remember correctly closing with X button is pretty brutal on Windows: it gives the process a small amount of time to clean up and then just kills it, which was often the case with cAER as it takes a few seconds to shutdown everything cleanly. If you don't shut the configuration TCP socket down correctly, it will still mark it as occupied for a certain time (until TCP timeouts expire), so if you restart cAER quickly it will fail thinking the address is already in use.
That's why I disabled the red X button, to force people to use CTRL+C, which does a clean shut down.
I'm not sure why it shows up for you, it should be greyed out... Maybe Windows 10 does things differently there.

federicohyo, 2017-03-12:
I can confirm that the closing button is disabled on windows 10 when running caer-bin by double-clicking o n it. The error I saw happens when running caer with mingw64 and then (erroneously) closing mingw64 by pressing the X button terminates the program in a bad status.