inivation/caer #52: recording data with davis640?

Thanh-Binh, 2017-03-20:
Currently, I use the following option for recording data:
cmake -DDAVISFX3=1 -DENABLE_FILE_OUTPUT=1 -DENABLE_BAFILTER=1 -DENABLE_VISUALIZER=1 . && make -j8 

and my caer-config.xml looks like
<node name="7-FileOutput" path="/1/7-FileOutput/">
                <attr key="bufferMaxInterval" type="int">20000</attr>
                <attr key="bufferSize" type="int">16384</attr>
                <attr key="directory" type="string">/home/binh/Neuromorphics/data</attr>
                <attr key="keepPackets" type="bool">true</attr>
                <attr key="prefix" type="string">caerOut</attr>
                <attr key="runAtStartup" type="bool">true</attr>
                <attr key="transferBufferSize" type="int">128</attr>
                <attr key="validOnly" type="bool">false</attr>
            </node>

But I really do not find out any aedat file in this folder.
Do anybody know where is that problem?
Thanks.

llongi, 2017-03-21:
This should work. Did you try removing the CMake cache like Federico suggested in the other ticket?

Thanh-Binh, 2017-03-21:
it works now so I will close this issue, best thank

closed 2017-03-21 by Thanh-Binh