inivation/caer #57: false timestamp

Thanh-Binh, 2017-03-29:
Hi all,
today, I look at the timestamp of the events (using caerPolarityEventGetTimestamp) received by clients over network, but they are not understandable (sometimes not incremental, sometimes negativ).
Is there any bug here in the current version?

t=: 1729362523; n= 5224
t=: 1076789692; n= 4870
t=: 1688647470; n= 5152
t=: 769035397; n= 4668
t=: 1281869559; n= 4899
t=: 1158919196; n= 4815
t=: 1381041222; n= 4900
t=: -1685764502; n= 5477
t=: -101140578; n= 4131
t=: 656753164; n= 4479
t=: -527140652; n= 3886
t=: -2032407800; n= 5207
t=: -1838782045; n= 3223
t=: -579540490; n= 3807
t=: 1752974291; n= 2860
t=: -223190078; n= 3939
t=: 1948740790; n= 2925
t=: -988227462; n= 3546

llongi, 2017-03-29:
I'm not aware of any current problems. You should never get negative timestamps, I think you're decoding the data wrong. Can you show the code responsible for decoding and printing the above output? What network module are you using, TCP?

Thanh-Binh, 2017-03-29:
Maybe decoding problem..
Where can I look at? How can I know network module I used? I compile like
cmake -DDAVISFX3=1 -DDAVISFX2=0 -DENABLE_BAFILTER=1 -DENABLE_STATISTICS=1 -DENABLE_VISUALIZER=1 -DENABLE_NETWORK_OUTPUT=1 . && make -j8

Thanh-Binh, 2017-03-29:
@llongi I use uv_os_sock_t socket.
@breznak Do you have the same problem?

breznak, 2017-03-29:
@Thanh-Binh no, send/receiving over net works well for me. But I had similar "bugs", I'm almost certain it's a bug in decoding the data..

Thanh-Binh, 2017-03-29:
@breznak could you please solve this problem?

llongi, 2017-04-07:
Closing this, seems to be an issue on your client's side.
Remember that the network stream has also an initial 20 byte header, then it's just event packets as usual.

closed 2017-04-07 by llongi