inivation/caer #59: support for offline datasets

alirezadir, 2017-04-10:
Hello, I was wondering if the current version of the framework can be used to work with offline dataset (e.g. davis dataset). Thanks.

llongi, 2017-04-11:
We do have an input module that can read files in the AEDAT 3.1 format, which is what the output modules of cAER produce.
What kind of datasets do you want to open? From where did you get them?

alirezadir, 2017-04-11:
Thanks, I'd like to work with the open source dataset from uzh (event-camera dataset and simulator). Their events data is in text file format with (timestamp x y polarity) per line. What would be the best way to read such data?

llongi, 2017-04-12:
cAER doesn't support that format as input, though it could be added.
But do you need/want to use cAER, or do you simply want to work with that data to test things? Then maybe just write a function that reads the text line-by-line and parses it using your preferred language, i.e. Python or Matlab.