inivation/caer #61: Documentation for modules config parameters?

breznak, 2017-04-19:
Is there a doc for all the parameters used in `caer-config.xml` ? 

I was recently searching for the exact meaning for `bufferMaxInterval`, and couldn't grep it in the source codes...

llongi, 2017-04-19:
That's because 'bufferMaxInterval' doesn't exist (anymore). It must have been some old setting that got renamed or removed. The configuration system currently doesn't support removing nodes/attributes from it, as it would be pretty difficult to track those things properly. In the next version of cAER with modules support, this is slightly helped by importing all the settings as READ_ONLY, and only activating settings that actually exist. So you could then write some form of cleanup by looking at which things are left READ_ONLY. Also the various GUIs (CLI, JavaFX) will deny changed to such parameters.

llongi, 2017-04-19:
Btw, I think the one you're referring to is now 'PacketContainerInterval'. And that is documented in libcaer at http://inilabs.github.io/libcaer/html/usb_8h.html#add13e8c8848d7fbc850ff9b17e8998f6

breznak, 2017-04-19:
> That's because 'bufferMaxInterval' doesn't exist (anymore).  

Oh, good to know! Looks like the old config I've been carrying around will need some spring cleanup! 

Maybe we should spill a warning message on unused parameters in the module's config, I think that would be enough. 

Thank you again! 

PS: I think it may not be a PacketContainerInterval, as was related to Network output, but the doc is very nice, I'll search there. 
PPS: maybe adding a link to this doc in the config.xml?

llongi, 2017-04-19:
Well the doc there is for libcaer and only helps for device-related parameters.
For other modules like the I/O ones, BAFilter etc. there is indeed no formal documentation (yet).
Output modules have no interval-related configuration, Input modules to have a 'PacketContainerInterval'.
Most modules currently have a minimal doc as a comment right next to the configuration parameter declaration. It might be a good idea to make documentation part of the interface. In the new modules branch we separate creation of configuration parameters from their usage, and you can now specify ranges, as well as flags such as READ_ONLY etc. Maybe it would make sense to require a documentation string to be added, so that the various interfaces (CLI, JavaFX GUI etc.) can easily retrieve a tool-tip-like small documentation for each parameter.

llongi, 2017-04-19:
I'll keep this open until I've implemented something like that or written proper documentation, to track that.

llongi, 2017-05-31:
All sshsNodeCreate() calls in the modules branch now have an ulterior parameters: a string (const char *) that describes the attribute's function. This can be queried via the config-server (CAER_CONFIG_GET_DESCRIPTION) and will be used in caer-ctl to provide a 'help' command, and in caerctl-gui-javafx to add tooltips.

closed 2017-05-31 by llongi