inivation/caer #65: Any document about the new filter?

Thanh-Binh, 2017-04-20:
do you have any document or any information / hint about the effect of the new filters like Denoisy AutoEncoder, Gesture learning filter, learning filter, meanrate filter etc.?

How can I get the results/outputs of each filter?
Thanks

federicohyo, 2017-04-20:
Hi Binh,

unfortunately these new filters are highly experimentals and will not be supported by us in the next caer release (that will be in about a week). However, they might get supported in the upcoming months. In addition, these filters are used in conjuction with the Dynap-se neuromorphic processor and without such a processor you will not able to run them.

with best regards,
 Federico

closed 2017-04-20 by llongi

Thanh-Binh, 2017-04-20:
@federicohyo @llongi do you have any filter for filtering events generated by camera motion?