inivation/caer #75: Poisson type stimulation Dynap-se

NatINI, 2017-05-09:
Hi,

I am at the moment trying to code in order to have Poisson type stimulation on the Dynap-se. `
Following is the piece of code related to a regular stimulation:

if (atomic_load(&state->genSpikeState.stim_duration)
		<= ((double) tend.tv_sec + 1.0e-9 * tend.tv_nsec) - ((double) tstart.tv_sec + 1.0e-9 * tstart.tv_nsec)) {
		if (atomic_load(&state->genSpikeState.started)) {

Can anyone tell me how to change this in order to have a poisson type stimulation (i.e. Gaussian response)

Cheers

federicohyo, 2017-05-09:
Hi Nat,

to generate a Poisson spike train you will need to generate inter-spike-intervals (ISI) distributed accordingly to a Poisson distribution.
You can easily do that by picking a random number and then calculating the next time of occurrence of a spike as:

def nextTime(rate):
    return -math.log(1.0 - random.random()) / rate

in which the rate parameter is the mean rate of your poisson spike train. random.random is a random number between 0 and 1.

The bit of code that you can use as an inspiration for the Poisson generator is the one -> https://github.com/inilabs/caer/blob/master/modules/ini/gen_spikes.c  , lines 438/500. 
In there, we have a basic regular spike train implementation. Just make a function like that one which uses tstart and tend structures for knowing what is the current time and when is the last time what you have sent a spike.

In addition, I think that Dongchen has already implemented it. So you could also try to ask for his help in solving this task.

best,
 Federico

llongi, 2017-06-01:
See issue #79.

llongi, 2017-06-06:
Merged pull request #79, thanks to @clauniel for working on this.

closed 2017-06-06 by llongi