inivation/caer #76: compilation problem by current version

Thanh-Binh, 2017-05-11:
I have just clone both libcaer and caer.
I successfully installed libcaer using sudo and libcaer is installed in /usr/lib/x86_64-linux-gnu/
lrwxrwxrwx 1 root root     12 Aug 29  2016 /usr/lib/x86_64-linux-gnu/libcaer.so -> libcaer.so.2
lrwxrwxrwx 1 root root     16 May 11 14:11 /usr/lib/x86_64-linux-gnu/libcaer.so.2 -> libcaer.so.2.0.2
-rw-r--r-- 1 root root 154672 Feb  2 09:24 /usr/lib/x86_64-linux-gnu/libcaer.so.2.0.0
-rw-r--r-- 1 root root 135472 Mar 21 12:44 /usr/lib/x86_64-linux-gnu/libcaer.so.2.0.1
-rw-r--r-- 1 root root 139816 May 11 14:36 /usr/lib/x86_64-linux-gnu/libcaer.so.2.0.2

By installing caer, got the following errror with check for libcaer version. It is 1st time I have update problem. I really do not understand. Could anyone help me? Thanks

cmake -DDAVISFX3=1 -DDAVISFX2=0 -DENABLE_BAFILTER=1 -DENABLE_STATISTICS=1 -DENABLE_VISUALIZER=1 -DENABLE_MEDIANTRACKER=1 .

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Check size of size_t
-- Check size of size_t - done
-- Check size of void *
-- Check size of void * - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libcaer>=2.0.1'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  CMakeLists.txt:174 (PKG_CHECK_MODULES)

-- Configuring incomplete, errors occurred!

breznak, 2017-05-11:
> -rw-r--r-- 1 root root 154672 Feb 2 09:24 /usr/lib/x86_64-linux-gnu/libcaer.so.2.0.0
-rw-r--r-- 1 root root 135472 Mar 21 12:44 /usr/lib/x86_64-linux-gnu/libcaer.so.2.0.1
-rw-r--r-- 1 root root 139816 May 11 14:36 /usr/lib/x86_64-linux-gnu/libcaer.so.2.0.2

You have multiple (old) version of the lib lying around, this may cause the problem. 
- manually remove them
- also remove pkgconfig generated by libcaer
- rebuild libcaer

Thanh-Binh, 2017-05-11:
if I comment this line and insert
set (LIBCAER_INCLUDE_DIRS "/usr/include/libcaer")
set (LIBCAER_LIBRARY_DIRS "/usr/lib/x86_64-linux-gnu/")
set (LIBCAER_LIBRARIES "/usr/lib/x86_64-linux-gnu/libcaer.so")
it works...

Thanh-Binh, 2017-05-11:
It is very nice if it supports
Sudo make uninstall
So that we can remove all lib an header files of unused version. What do yo think?
How can I remove pkg-config generated from libcaer?

breznak, 2017-05-11:
> Sudo make uninstall
So that we can remove all lib an header files of unused version. What do yo think?

https://github.com/inilabs/libcaer/issues/19

llongi, 2017-05-16:
Having multiple lib files around cannot be the issue, that is perfectly normal way to do things on Linux and will not confuse the linker/loader. Here you're seeing pkg-config being unable to find libcaer, so to me it looks like the libcaer pkg-config file (libcaer.pc) is not installed or installed somewhere that your pkg-config install cannot find it. Usually it should be in $LIBDIR/pkgconfig/libcaer.pc, so check if  /usr/lib/x86_64-linux-gnu/pkgconfig/libcaer.pc exists, and verify that that is indeed a directory where your pkg-config is searching for its files. If the files is in the right place and your pkg-config is working, then the following command "pkg-config --libs libcaer" should output "-lcaer".

Thanh-Binh, 2017-05-16:
It is exact what we require that  libcaer installation should put libcaer.pc into the right folder so that other sw like caer can find it by configuring using cmake. Unfortunately it is not the case with he current version!

breznak, 2017-05-16:
> Unfortunately it is not the case with he current version!

Where is the file for you? It is working correctly for me

Thanh-Binh, 2017-05-23:
I have just clone the current version of libcaer and installed it successfully.
However, I CAN not compile caer.
Here is the following errors:

/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libcaer.so: undefined reference to `powf'
collect2: error: ld returned 1 exit status
/usr/lib/gcc/x86_64-linux-gnu/5/../../../x86_64-linux-gnu/libcaer.so: undefined reference to `powf'
collect2: error: ld returned 1 exit status
utils/tcpststat/CMakeFiles/tcpststat.dir/build.make:95: recipe for target 'utils/tcpststat/tcpststat' failed
make[2]: *** [utils/tcpststat/tcpststat] Error 1
CMakeFiles/Makefile2:1071: recipe for target 'utils/tcpststat/CMakeFiles/tcpststat.dir/all' failed
make[1]: *** [utils/tcpststat/CMakeFiles/tcpststat.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
utils/udpststat/CMakeFiles/udpststat.dir/build.make:95: recipe for target 'utils/udpststat/udpststat' failed
make[2]: *** [utils/udpststat/udpststat] Error 1
CMakeFiles/Makefile2:1126: recipe for target 'utils/udpststat/CMakeFiles/udpststat.dir/all' failed
make[1]: *** [utils/udpststat/CMakeFiles/udpststat.dir/all] Error 2

Can you help me?

Thanh-Binh, 2017-05-23:
After clone the current caer, and compiling, I have the following errors:
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Check size of size_t
-- Check size of size_t - done
-- Check size of void *
-- Check size of void * - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libcaer>=2.0.1'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  CMakeLists.txt:174 (PKG_CHECK_MODULES)

-- Configuring incomplete, errors occurred!
See also "/home/binh/Neuromorphics/caer/CMakeFiles/CMakeOutput.log".
See also "/home/binh/Neuromorphics/caer/CMakeFiles/CMakeError.log".

That is the same problem I described 12 days ago...
It is NOT solved....

llongi, 2017-05-23:
The 'undefined reference to powf' problem should be solved in libcaer now, I forgot to explicitly link to libm for the math.h functions, and most systems to do that implicitly so it doesn't always show up.
Regarding this other problem, it's clearly a pkg-config error: your pkg-config cannot find your libcaer installation. This means either your pkg-config is broken in some way, or your libcaer installation ended up somewhere that pkg-config doesn't search for. What's the output of 'pkg-config --version' and 'pkg-config --debug' ? Please paste them here.

Thanh-Binh, 2017-05-23:
Did you update libcaer? Thanks
Here are the output you requested:
pkg-config --version
0.29.1

 pkg-config --debug
no output option set, defaulting to --exists
Error printing disabled by default due to use of output options --exists, --atleast/exact/max-version, --list-all or no output option at all. Value of --print-errors: 0
Error printing disabled
Adding virtual 'pkg-config' package to list of known packages
Cannot open directory #1 '/usr/local/lib/x86_64-linux-gnu/pkgconfig' in package search path: No such file or directory
Scanning directory #2 '/usr/local/lib/pkgconfig'
File 'allegro_acodec-5.0.pc' appears to be a .pc file
Will find package 'allegro_acodec-5.0' in file '/usr/local/lib/pkgconfig/allegro_acodec-5.0.pc'
File 'allegro_memfile-5.0.pc' appears to be a .pc file
Will find package 'allegro_memfile-5.0' in file '/usr/local/lib/pkgconfig/allegro_memfile-5.0.pc'
File 'mxml.pc' appears to be a .pc file
Will find package 'mxml' in file '/usr/local/lib/pkgconfig/mxml.pc'
File 'allegro_ttf-5.0.pc' appears to be a .pc file
Will find package 'allegro_ttf-5.0' in file '/usr/local/lib/pkgconfig/allegro_ttf-5.0.pc'
File 'allegro_dialog-5.pc' appears to be a .pc file
Will find package 'allegro_dialog-5' in file '/usr/local/lib/pkgconfig/allegro_dialog-5.pc'
File 'allegro_audio-5.0.pc' appears to be a .pc file
Will find package 'allegro_audio-5.0' in file '/usr/local/lib/pkgconfig/allegro_audio-5.0.pc'
File 'allegro_primitives-5.pc' appears to be a .pc file
Will find package 'allegro_primitives-5' in file '/usr/local/lib/pkgconfig/allegro_primitives-5.pc'
File 'allegro_color-5.0.pc' appears to be a .pc file
Will find package 'allegro_color-5.0' in file '/usr/local/lib/pkgconfig/allegro_color-5.0.pc'
File 'allegro-5.pc' appears to be a .pc file
Will find package 'allegro-5' in file '/usr/local/lib/pkgconfig/allegro-5.pc'
File 'libzmq.pc' appears to be a .pc file
Will find package 'libzmq' in file '/usr/local/lib/pkgconfig/libzmq.pc'
File 'allegro_audio-5.pc' appears to be a .pc file
Will find package 'allegro_audio-5' in file '/usr/local/lib/pkgconfig/allegro_audio-5.pc'
File 'libcaer.pc' appears to be a .pc file
Will find package 'libcaer' in file '/usr/local/lib/pkgconfig/libcaer.pc'
File 'glfw3.pc' appears to be a .pc file
Will find package 'glfw3' in file '/usr/local/lib/pkgconfig/glfw3.pc'
File 'libfreenect.pc' appears to be a .pc file
Will find package 'libfreenect' in file '/usr/local/lib/pkgconfig/libfreenect.pc'
File 'allegro_color-5.pc' appears to be a .pc file
Will find package 'allegro_color-5' in file '/usr/local/lib/pkgconfig/allegro_color-5.pc'
File 'allegro_image-5.0.pc' appears to be a .pc file
Will find package 'allegro_image-5.0' in file '/usr/local/lib/pkgconfig/allegro_image-5.0.pc'
File 'allegro_font-5.pc' appears to be a .pc file
Will find package 'allegro_font-5' in file '/usr/local/lib/pkgconfig/allegro_font-5.pc'
File 'libusb-1.0.pc' appears to be a .pc file
Will find package 'libusb-1.0' in file '/usr/local/lib/pkgconfig/libusb-1.0.pc'
File 'allegro_image-5.pc' appears to be a .pc file
Will find package 'allegro_image-5' in file '/usr/local/lib/pkgconfig/allegro_image-5.pc'
File 'allegro-5.0.pc' appears to be a .pc file
Will find package 'allegro-5.0' in file '/usr/local/lib/pkgconfig/allegro-5.0.pc'
File 'allegro_main-5.0.pc' appears to be a .pc file
Will find package 'allegro_main-5.0' in file '/usr/local/lib/pkgconfig/allegro_main-5.0.pc'
File 'allegro_ttf-5.pc' appears to be a .pc file
Will find package 'allegro_ttf-5' in file '/usr/local/lib/pkgconfig/allegro_ttf-5.pc'
File 'allegro_memfile-5.pc' appears to be a .pc file
Will find package 'allegro_memfile-5' in file '/usr/local/lib/pkgconfig/allegro_memfile-5.pc'
File 'allegro_font-5.0.pc' appears to be a .pc file
Will find package 'allegro_font-5.0' in file '/usr/local/lib/pkgconfig/allegro_font-5.0.pc'
File 'allegro_primitives-5.0.pc' appears to be a .pc file
Will find package 'allegro_primitives-5.0' in file '/usr/local/lib/pkgconfig/allegro_primitives-5.0.pc'
File 'opencv.pc' appears to be a .pc file
Will find package 'opencv' in file '/usr/local/lib/pkgconfig/opencv.pc'
File 'allegro_dialog-5.0.pc' appears to be a .pc file
Will find package 'allegro_dialog-5.0' in file '/usr/local/lib/pkgconfig/allegro_dialog-5.0.pc'
File 'allegro_main-5.pc' appears to be a .pc file
Will find package 'allegro_main-5' in file '/usr/local/lib/pkgconfig/allegro_main-5.pc'
File 'allegro_acodec-5.pc' appears to be a .pc file
Will find package 'allegro_acodec-5' in file '/usr/local/lib/pkgconfig/allegro_acodec-5.pc'
[... 512 lines cut ...]
File 'xtrans.pc' appears to be a .pc file
Will find package 'xtrans' in file '/usr/share/pkgconfig/xtrans.pc'
File 'dri2proto.pc' appears to be a .pc file
Will find package 'dri2proto' in file '/usr/share/pkgconfig/dri2proto.pc'
File 'adwaita-icon-theme.pc' appears to be a .pc file
Will find package 'adwaita-icon-theme' in file '/usr/share/pkgconfig/adwaita-icon-theme.pc'
File 'renderproto.pc' appears to be a .pc file
Will find package 'renderproto' in file '/usr/share/pkgconfig/renderproto.pc'
File 'xproto.pc' appears to be a .pc file
Will find package 'xproto' in file '/usr/share/pkgconfig/xproto.pc'
File 'fontutil.pc' appears to be a .pc file
Will find package 'fontutil' in file '/usr/share/pkgconfig/fontutil.pc'
File 'xineramaproto.pc' appears to be a .pc file
Will find package 'xineramaproto' in file '/usr/share/pkgconfig/xineramaproto.pc'
File 'eigen3.pc' appears to be a .pc file
[... cut ...]
Will find package 'compositeproto' in file '/usr/share/pkgconfig/compositeproto.pc'
File 'mobile-broadband-provider-info.pc' appears to be a .pc file
Will find package 'mobile-broadband-provider-info' in file '/usr/share/pkgconfig/mobile-broadband-provider-info.pc'
File 'xf86vidmodeproto.pc' appears to be a .pc file
Will find package 'xf86vidmodeproto' in file '/usr/share/pkgconfig/xf86vidmodeproto.pc'
File 'xbitmaps.pc' appears to be a .pc file
Will find package 'xbitmaps' in file '/usr/share/pkgconfig/xbitmaps.pc'
Must specify package names on the command line

llongi, 2017-05-24:
If you search for libcaer in that output, it appears twice:

- first time when searching in /usr/local/lib/pkgconfig/:

File 'libcaer.pc' appears to be a .pc file
Will find package 'libcaer' in file '/usr/local/lib/pkgconfig/libcaer.pc'

- second time when searching in /usr/lib/x86_64-linux-gnu/pkgconfig/:

File 'libcaer.pc' appears to be a .pc file
File 'libcaer.pc' ignored, we already know about package 'libcaer'

So you have a second, older libcaer install in /usr/local/lib/, which is hiding the new, correct install in /usr/lib/x86_64-linux-gnu/pkgconfig/, and that's why this is failing. Clean out the old install and you'll be fine. Maybe you already did and just forgot to also remove that old pkg-config file.
Closing this bug.

closed 2017-05-24 by llongi

Thanh-Binh, 2017-05-24:
Thanks for your support! As mentioned sometimes ago, i wish a possibility to clean the old version (lib, libcaer.pc etc.) before install the new one!

tobidelbruck, 2017-06-22:
https://github.com/inilabs/caer/issues/76 same as this issue probably. Try making sure there is no existing older libcaer in /usr or /usr/local.