inivation/caer #77: Problem using C++ libcaercpp for net listening & cAER format parsing

breznak, 2017-05-11:
Basically, I'm trying to rewrite `./utils/tcpststat/tcpststat.c` into C++ using the `libcaercpp/` C++ headers to achieve that. 

I used to have the code working using the C-headers `libcaer/`, but C++ probably fixes my weird problem C/C++ linker problems. 

I'm able to compile, but the code produces wrong output at runtime. 

PS: Would it be possible to integrate more networking support into libcaer? 

The code
```
// this file shows network I/O functionality and cAER parsing
...
#include <libcaercpp/libcaer.hpp> //libcaer; use -lcaer
#include <libcaercpp/network.hpp>
#include <libcaercpp/events/common.hpp>
#include <libcaercpp/events/polarity.hpp>
#include <libcaercpp/events/frame.hpp>

#include <iostream>
#include <cassert>

using xxx::network::uv::Net; //UV related low level methods: rcvAll,...
using namespace xxx::utils::network;
using libcaer::network::AEDAT3NetworkHeader;
using libcaer::events::EventPacket;
using libcaer::events::PolarityEventPacket;
using libcaer::events::FrameEventPacket;

int DEBUG = 4;

int main ()
{

  //enable debug print outs
   size_t iter=0;

 //  Prepare our context and socket
  Net net("127.0.0.1", 7777); //connects to port

// STEP 1: Get the 1st packet - Header (20B)
    auto d = net.rcvAll(AEDAT3_NETWORK_HEADER_LENGTH);
      AEDAT3NetworkHeader h(d.data());
    assert(h.checkVersionNumber() && h.checkMagicNumber()); //FIXME
cout << "Net header \t ...OK" <<endl;

    //  Do 10 requests, waiting each time for a response
    while (iter <1000000) {//1M
        // STEP 2: parse packet header to compute following "data size"
    auto packet = net.rcvAll(CAER_EVENT_PACKET_HEADER_SIZE);
    d.insert(end(d), begin(packet), end(packet));
    auto header =(caerEventPacketHeader)&packet; //How do I construct the header in C++?

//below is all debug
if(DEBUG >=2) {
                int16_t eventType = header->eventType;
                int16_t eventSource = header->eventSource;
                int32_t eventSize = header->eventSize;
                int32_t eventTSOffset = header->eventTSOffset;
                int32_t eventTSOverflow = header->eventTSOverflow;
                int32_t eventCapacity = header->eventCapacity;
                int32_t eventNumber = header->eventNumber;
                int32_t eventValid = header->eventValid;

cout << "type = " << eventType << ", source = " << eventSource << ", size = " << eventSize << ", tsOffset = " << eventTSOffset << ", tsOverflow = "
  << eventTSOverflow<< ", capacity = " << eventCapacity << ", number = " << eventNumber <<  ", valid = " << eventValid << endl;
}
if(header->eventType != POLARITY_EVENT || header->eventSize < 0 || header->eventCapacity<0) {
  cout << "wrong data" << endl;
  continue;
}
    size_t DATA_SIZE = header->eventSize * header->eventCapacity;
    assert(header->eventSize > 0 && header->eventCapacity > 0  );

    // append packet header + data to form a full packet
    auto data = net.rcvAll(DATA_SIZE);
    packet.insert(end(packet), begin(data), end(data));

// STEP 3: parse polarity packets
    if (header->eventType== POLARITY_EVENT) {
      PolarityEventPacket polarityPacket = PolarityEventPacket((caerEventPacketHeader) packet.data());

  for(auto evt : polarityPacket) {
    if(!evt.isValid()) { continue;} // skip invalid events
    //construct struct to hold new values
    PolarityPoint pt(evt.getTimestamp(), evt.getX(), evt.getY(), evt.getPolarity());
    assert(pt.ts > 0 && pt.x <= DAVIS_240B_SIZE_X && pt.y <= DAVIS_240B_SIZE_Y);

    if(DEBUG >=3) cout << "Polarity event data - X Address: \t" << pt.x << ", Y Address \t" << pt.y << ", Polarity: \t"<<pt.p << " , TS \t" << pt.ts << endl;

  } //end-for

// STEP 3: (..or) parse Frame packets (APS) 
  } else if (header->eventType == FRAME_EVENT) {
  std::vector<FramePoint> frames;

  FrameEventPacket framePacket = FrameEventPacket((caerFrameEventPacket)packet.data());

  for(auto evt : framePacket) {

    FramePoint frame(evt.getLengthX(),evt.getLengthY(), evt.getPixelArrayUnsafe());
    frames.emplace_back(frame); //do sth with frame

    if(DEBUG >=3) cout <<"First frame event data - X Size: " <<frame.frameSizeX << ", Y Size: " <<frame.frameSizeY << ", Pixel 0 Value: " <<frame.framePixels[0]<< endl;
  }
 }    
 } //while
return 0;
}

```

And the code in `Net` is : 

```
#include <xxx/utils/network/lib_net_uv.hpp>

#include <iostream>
// uncomment to disable assert()
// #define NDEBUG
#include <cassert>
#include <thread>
#include <chrono>

using namespace std;
using namespace xxx::utils::network::uv;

//forward declarations
bool recvUntilDone(sock_t, uint8_t*, size_t);

// implementation
Net::Net(string URI, unsigned int port):
  URI_(URI), port_(port) {
  //init socket
  // Create listening socket for TCP data.
  this->socket_ = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
   assert(socket_ >=0);
  conn(); //TODO the conn() has to be here otherwise first 20B parsing fails; fix that
}

void Net::conn(string URI, unsigned int port) {
  //  Prepare our context and socket
  struct sockaddr_in listenTCPAddress;
  auto rc = uv_ip4_addr(URI.c_str(), (uint16_t)port, &listenTCPAddress);
    assert(rc==0);
  int rd = -1;
    do {
  cout << "Attempting to connect..." << endl;
  std::this_thread::sleep_for(std::chrono::milliseconds(1000));
  rd = connect(this->socket_, (struct sockaddr *) &listenTCPAddress, sizeof(struct sockaddr_in));
  } while(rd < 0); 
  assert(rd>=0);
  std::cout << "connected"<<std::endl;
}

void Net::conn() {
  conn(this->URI_, this->port_);
}

/** reveive a (multi-part) message completely */
vect_t Net::rcvAll(size_t bytesToRead) {
assert(bytesToRead >0);
  this->buffer_.resize(bytesToRead);// the vector MUST be initialized to the size, as we treat it as an array for uv's recvUntilDone() 
assert(buffer_.size()==bytesToRead);
assert(this->socket_);
  auto rc=  recvUntilDone(this->socket_, this->buffer_.data(), bytesToRead);
assert(rc);

  return buffer_;
}

//private methods

/**
 * Read N bytes from the socket S into buffer B.
 *
 * @param sock socket S.
 * @param buffer buffer B.
 * @param bytesToRead number of bytes to read.
 *
 * @return Return true on success, false on failure.
 */
bool recvUntilDone(sock_t sock, uint8_t *buffer, size_t bytesToRead) {
  size_t curRead = 0;

  while (curRead < bytesToRead) {
    auto rc = recv(sock, buffer + curRead, bytesToRead - curRead, 0);
    if (rc <= 0) return false;
    curRead += (size_t) rc;
  }
  return true;
}
```

breznak, 2017-05-11:
- What parsing I'm I doint wrong if I want to use the C++-only headers? 
- Would you consider something like the class `Net` in libcaer to simplify the network I/O? Then eg. the `utils/` in caer would take advantage of it (currently using the C-code).

breznak, 2017-05-11:
This is the problem I'm trying to solve: https://github.com/inilabs/libcaer/issues/14

Thanh-Binh, 2017-05-11:
I find our full c++ version very helpfull, it avoid any mixing ansi-c and c++ at the client side. Pls share it if your bugs are fixed

breznak, 2017-05-15:
@llongi @federicohyo Please any solution to the better C++ use?

llongi, 2017-05-16:
So, a few things here:
1) Even if you decide a packet is not interesting (ie when you check that if packet != POLARITY_EVENT then continue), you still must read the data of the current packet (read(eventSize*eventCapacity)) always.
2) Constructing a C++ EventPacket by passing in the raw data pointer is fine. What you might want to check is the second possible parameter to the constructor there: takeMemoryOwnership. By default this is true, meaing the C++ EventPacket takes over memory management for that data pointer you passed to the constructor and will try to free() it when the EventPacket goes out of scope (when the destructor is called). If you are already managing that memory yourself, which in this case I think happens by "packet" going out of scope, you need to set that parameter to false, so that the EventPacket doesn't try to free the memory twice.

breznak, 2017-05-16:
> Even if you decide a packet is not interesting (ie when you check that if packet != POLARITY_EVENT then continue), you still must read the data of the current packet (read(eventSize*eventCapacity)) always. 

Thank you very much, @llongi ! This was extremely helpful! 

Some follow up questions: 
1. should the Packet header be constructed with the NetworkHeader data, or without it? 

>        // STEP 2: parse packet header to compute following "data size"
    auto packet = net.rcvAll(CAER_EVENT_PACKET_HEADER_SIZE);
    d.insert(end(d), begin(packet), end(packet));
    auto header =(caerEventPacketHeader)&packet; //How do I construct the header in C++?

vs 

>     auto header =(caerEventPacketHeader)&d; //How do I construct the header in C++?
 
2. are there checks for PacketHeader? 

The same way as it exists for NetworkHeader

>       AEDAT3NetworkHeader h(d.data());
    assert(h.checkVersionNumber() && h.checkMagicNumber()); //FIXME

I'm getting corruption of the PacketHeader (below), so it would be nice to call `packetH.check()` that would verify sane values.

Could you also take a look on the code during connection, please? I'm getting corrupted (again seems like wrong order of reading the stream) PacketHeader: 

```
Attempting to connect...
Attempting to connect...
Attempting to connect...
connected
Net header       ...OK
type = -20944, source = 140, size = 0, tsOffset = 9219680, tsOverflow = 0, capacity = 9219680, number = 0, valid = 6357408
caer_demo: xxx/caer_demo.cpp:60: int main(): Assertion `header->eventSize > 0 && header->eventCapacity > 0' failed.
``` 

PS: Would a library that houses all these checks and raw network stream reading be useful for caer / libcaer? 
```
Net net("127.0.0.1", 8080);
net.connect(); //connect, read NetworkHeader and check it
GenericEventPacket packet = net.readPacket(); //read next packet in the stream, do checks
```

llongi, 2017-05-16:
1) Without, the event packet is always only made up of the 28-byte packet header and the packet data (of size eventSize * eventCapacity).

2) Not really, all of its values can theoretically be any value, you could have a type with ID 32000 or event size 1GB. The network streams are guaranteed to always be 20-byte network header, followed by packets, which are made as I described in (1).

3) Such a library might be useful, some parts could even be integrated in caer/libcaer, but especially the networking code probably makes sense to keep separate, as everybody uses their own preferred library for that. Just publish it as a GitHub project. :-)

breznak, 2017-05-16:
> 1. 

Thank you for explanation! 
Then I'm lost, because the code seems correct on the first glance :( 
I've added some debug, and I'm still getting

```
Attempting to connect...
connected
Magic number: 2105305046418351704 Sequence number: 0 Version number: 1 Format number: 0 Source ID: 10
Net header       ...OK
type = 23984, source = 376, size = 0, tsOffset = 24665548, tsOverflow = 0, capacity = 24665548, number = 0, valid = 194668120
caer_demo: xxx/caer_demo.cpp:68: int main(): Assertion `header->eventSize > 0 && header->eventCapacity > 0
```

Which leaves room for mistake in the parsing
>  auto header =(caerEventPacketHeader)&packet; 

......
HURRAY!! :tada:  Yes, the error was in there, 
Instead of 
>  auto header =(caerEventPacketHeader)&packet; 

It needs to be `auto header =(caerEventPacketHeader)packet.data(); `  //(1)

Then the output is correct: 
```
type = 0, source = 10, size = 8, tsOffset = 4, tsOverflow = 0, capacity = 2, number = 2, valid = 2
type = 1, source = 10, size = 8, tsOffset = 4, tsOverflow = 0, capacity = 672, number = 672, valid = 672
type = 3, source = 10, size = 36, tsOffset = 4, tsOverflow = 0, capacity = 33, number = 33, valid = 33
type = 0, source = 10, size = 8, tsOffset = 4, tsOverflow = 0, capacity = 4, number = 4, valid = 4
```

> 2. Not really, all of its values can theoretically be any value,  

Understood, although, it would be useful to have some check, unless we have a library that does the I/O & parsing 100% correctly. 
We could have a list of known event types and check for validity with any of them. 

> 3. Such a library might be useful...

To avoid mistakes like (1), I'd like to propose a PR to make c++ libcaer events constructors such as `PolarityEvent(vector<uint8_t> rawData)` ..where: the parsing to caerHeader is done, some sanity checks can be done. What do you think?

breznak, 2017-05-16:
Thank you very much for help!

closed 2017-05-16 by breznak