inivation/caer #78: branch modules failed to compile

federicohyo, 2017-05-29:
I'm using the following cmake configuration

cmake -DDAVISFX3=1 -DDVS128=1 -DDAVISFX2=1 -DSTATISTICS=1 -DBAFILTER=1 -DVISUALIZER=1 .

the error I get is about libuv...
config_server.cpp:12:2: error: ‘uv_async_t’ does not name a type uv_async_t asyncShutdown;
config_server.cpp:16:36: error: variable or field ‘configServerConnection’ declared void static void configServerConnection(uv_stream_t *server, int status);
                                   
which version is required? I have libuv1-dev from ubuntu16.04 repo.. what am I missing?

thank you!

llongi, 2017-05-29:
Currently work in progress, it's normal that it's broken.

llongi, 2017-05-30:
Compiles fine again, config-server now uses Boost.ASIO.

closed 2017-05-30 by llongi