inivation/caer #80: linker command failed

NatINI, 2017-06-07:
Hi,

I just pulled caer and libcaer and the following error message appeared:

clang: error: linker command failed with exit code 1 (use -v to see invocation)

Any idea on this?

Cheers
Nat

llongi, 2017-06-07:
What are you doing when you get that error message? Did you execute cmake as described in the README before doing make? Did you install libcaer before trying to compile caer?

NatINI, 2017-06-07:
Yes I did all of this before getting that error message. I basically pulled
in order to get the poisson spike generator updated. Never saw that message
before.

llongi, 2017-06-07:
Try cleaning up the cmake temporary files:
`rm -Rf CMakeFiles/ CMakeCache.txt`
and then execute cmake + make again.

closed 2017-06-20 by llongi