inivation/caer #81: EventPacket Container

NatINI, 2017-06-16:
Hi,

I have been getting this error message when starting the dynap-se:

2017-06-16 09:56:11 (TZ+0200): NOTICE: Dynap-se ID-1 SN-00000009 [20:5]: Dropped EventPacket Container because ring-buffer full!

The problem is that this message will show up in a continuous loop not allowing me to run my program (helloworld).

Any ideas on this?

Cheers
Nat

federicohyo, 2017-06-16:
If you disable your helloworld module,  do you still get that error message?

federicohyo, 2017-06-19:
I agree this is an issue, sometimes it is needed to program the CAM and therefore

federicohyo, 2017-06-19:
This is a current issue of libcaer

NatINI, 2017-06-19:
Indeed in libcaer and dynapse.c the line associated with this "print" has to be commented in order to not see this print anymore. I am dropping many events during the initialisation and when this error message was happening it couldn't run my program.

closed 2017-06-22 by llongi