inivation/caer #82: HW (DavisFX240B) vs FileInput buffer-sizes problem

breznak, 2017-06-20:
Hello, I'm investigating a strange issue with streaming directly from the HW camera (="live") vs. when replaying from a file (="offline"). 

My setup is: hw Davis240B with the latest FW, latest upstream libcaer & caer. I use a custom application A that does multi-threading, parses the net I/O from caer and does my computations. 

A) in the "live" scenario, I stream directly from the camera over network and parse it in my app. The visualizer in cAER runs smooth, but my app "shutters", as it sometimes waits to receive data, sometimes the buffer is full. 

B) "offline", I record a scene with the camera (HW->File) and then replay the file (File->Network), all other settings are the same, but my app runs smooth. 

One would say the problem is in the multithreading& buffer management of my app, but there must be some change of how cAER groups & sends the data, as from the point-of-view of the app both scenarios are the same!

In the issue https://github.com/inilabs/caer/issues/61 I have asked about the buffer management in libcaer, and I take these are the relevant parameters for my scenario: 

```
                        <attr key="PacketContainerDelay" type="int">33000</attr>
                        <attr key="PacketContainerInterval" type="int">33000</attr>
                        <attr key="PacketContainerMaxPacketSize" type="int">1000</attr>
  
```

I have set them in `FileInput` config and all works fine (for "offline"), experimentally I try to set them to `NetworkOutput`, but does not seem to have an effect, I don't know here to set them for Davis ? 

From my `caer-config.xml` : 

* FileInput 

```
           <node name="1-FileInput" path="/1/1-FileInput/">
                <attr key="PacketContainerDelay" type="int">33000</attr>
                <attr key="PacketContainerInterval" type="int">33000</attr>
                <attr key="PacketContainerMaxPacketSize" type="int">1000</attr>
                <attr key="autoRestart" type="bool">false</attr>
                <attr key="bufferSize" type="int">65536</attr>
                <attr key="filePath" type="string">/tmp/a.aedat</attr>
                <attr key="keepPackets" type="bool">false</attr>
                <attr key="pause" type="bool">false</attr>
                <attr key="runAtStartup" type="bool">true</attr>
                <attr key="transferBufferSize" type="int">128</attr>
                <attr key="validOnly" type="bool">true</attr>
            </node>
```

* Network Output:  (I added the `PacketContainer*` but with no effect)

```
           <node name="9108-NetTCPServerOutput" path="/1/9108-NetTCPServerOutput/">
                <attr key="PacketContainerDelay" type="int">33000</attr>
                <attr key="PacketContainerInterval" type="int">33000</attr>
                <attr key="PacketContainerMaxPacketSize" type="int">1000</attr>
                <attr key="backlogSize" type="short">5</attr>
                <attr key="concurrentConnections" type="short">10</attr>
                <attr key="ipAddress" type="string">127.0.0.1</attr>
                <attr key="keepPackets" type="bool">false</attr>
                <attr key="portNumber" type="int">7777</attr>
                <attr key="ringBufferSize" type="int">128</attr>
                <attr key="runAtStartup" type="bool">true</attr>
                <attr key="validOnly" type="bool">true</attr>
            </node>
```

And the Davis (adds many settings) :  (I'm not sure if I added the params to DVS here, will test)

```
                   <node name="dvs" path="/1/1-DAVISFX2/DAVIS240B/dvs/">
                        <attr key="AckDelayColumn" type="byte">0</attr>
                        <attr key="AckDelayRow" type="byte">4</attr>
                        <attr key="AckExtensionColumn" type="byte">0</attr>
                        <attr key="AckExtensionRow" type="byte">1</attr>
                        <attr key="ExternalAERControl" type="bool">false</attr>
                        <attr key="FilterRowOnlyEvents" type="bool">true</attr>
                        <attr key="PacketContainerDelay" type="int">33000</attr>
                        <attr key="PacketContainerInterval" type="int">33000</attr>
                        <attr key="PacketContainerMaxPacketSize" type="int">1000</attr>
                        <attr key="Run" type="bool">true</attr>
                        <attr key="WaitOnTransferStall" type="bool">false</attr>
                    </node>
```

Or should I add to the `USB` ?

```
                    <node name="usb" path="/1/1-DAVISFX2/DAVIS240B/usb/">
                        <attr key="BufferNumber" type="int">8</attr>
                        <attr key="BufferSize" type="int">8192</attr>
                        <attr key="EarlyPacketDelay" type="short">8</attr>
                        <attr key="Run" type="bool">true</attr>
                    </node>
                </node>
```

Tried to change here, but with no luck ( I guess this is the control )
```
       <node name="system" path="/1/1-DAVISFX2/system/">
                    <attr key="DataExchangeBufferSize" type="int">64</attr>
                    <attr key="PacketContainerInterval" type="int">10000</attr>
                    <attr key="PacketContainerMaxPacketSize" type="int">8192</attr>
                </node>
```

Thank you!

llongi, 2017-06-20:
So, file input, that's correct how you set those three variables.
Network output does not have any such setting, the output always just writes out the packets as they come in from the input, no aggregation or resizing is done.
The DAVIS camera you need to set the two parameters "PacketContainerInterval" and "PacketContainerMaxPacketSize" in the "system" node (the last one you show), those indeed look like they are different from your file input setting, so just change them to match and check if the problem still exists.

breznak, 2017-06-20:
Thank you very much @llongi !! I'll report back in about 2 hrs when I can test this. 
(I think I did try with those values for System and wasn't successful, will retest!) 

> Network output does not have any such setting, the output always just writes out the packets as they come in from the input, no aggregation or resizing is done. 

So for the `Davis/dvs/` node those params are also ignored, and only the `System` is used for the real HW?

Thanh-Binh, 2017-06-20:
We have a lot of parameters for nodes, but I really do not have any overview which parameters for which nodes are valud or ignored? Is it possible to return errors if the parameters do not exist/ invalid value etc so that user will write a clean config file?

llongi, 2017-06-21:
Modules will create their own parameters and honor them, you never have to add things manually. So if it's in the file, it's fine, if it isn't, then it doesn't exist, so don't touch things manually. The new GUI + new modules branch will greatly simplify this.

breznak, 2017-06-21:
...I'm looking forward to the simplified filter config!! 
The feeling, when you spend 30 mins describing and analyzing your issue here in a comment, only to realize in the last sentence where the mistake was. :thought_balloon: :D 

I _think_ the suggested config helped the issue, closing till I confirm othervise. Thank you!

closed 2017-06-21 by breznak