inivation/caer #83: virgin make doesn't work, can't find libcaer

tobidelbruck, 2017-06-22:
I just git cloned libcaer and caer and got past sudo make install on libcaer on ubuntu 16. It installed to /usr/local/include/libcaaer.  
Then I tried to run example cmake on caer. I got this error

>  cmake -DDAVISFX2=1 -DENABLE_NETWORK_OUTPUT=1 -DENABLE_VISUALIZER=0 -DENABLE_STATISTICS=0 .
> -- Checking for module 'libcaer>=2.0.1'
> --   
> CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
>   A required package was not found
> Call Stack (most recent call first):
>   /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
>   CMakeLists.txt:173 (PKG_CHECK_MODULES)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/hyo/git/caer/CMakeFiles/CMakeOutput.log".
> See also "/home/hyo/git/caer/CMakeFiles/CMakeError.log".

tobidelbruck, 2017-06-22:
Actually cmake won't run initial cmake .

hyo@hyo-VirtualBox:~/git/caer$ cmake .
-- Checking for module 'libcaer>=2.0.1'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  CMakeLists.txt:173 (PKG_CHECK_MODULES)

The default install location of libcaer ends up installing to location that is not found in subsequent caer cmake.

tobidelbruck, 2017-06-22:
https://github.com/inilabs/caer/issues/76 related to same issue

tobidelbruck, 2017-06-22:
Solved by manually going through /usr and /usr/local and sudo rm all libcaer related files....

tobidelbruck, 2017-06-22:
And then rerunning the cmake above. It now works.

llongi, 2017-06-22:
Ok, yes, systems need to be kept consistent. :-)
See my reply in inilabs/libcaer#29.

closed 2017-06-22 by llongi