inivation/caer #85: A default caer-config.xml?

Thanh-Binh, 2017-06-30:
It is very nice for user to have a default config file from developers because we do not have any overview about which parameters are valid and can be changed?
Thanks

llongi, 2017-07-05:
With the newly merged modules system in master, there now is docs/davis-config.xml and docs/dynapse-config.xml to start with. By default cAER generates an empty config file, and when you add modules to it they add their own configuration parameters, so if they exist they are valid. Further if a parameter is now removed from a module, it will not save it again to the XML file at the end of the run, so you don't get "stale" parameters hanging around.

closed 2017-07-05 by llongi

Thanh-Binh, 2017-07-05:
Thanks for your explain