inivation/caer #97: using file input for visualizer hangs cAER

clauniel, 2017-07-10:
Using the config file below cAER shows a blank visualizer window when setting an input file name and hangs forever.

<sshs version="1.0">
    <node name="" path="/">
        <node name="FileInput" path="/FileInput/">
            <attr key="logLevel" type="byte">5</attr>
            <attr key="moduleId" type="short">5</attr>
            <attr key="moduleLibrary" type="string">caer_input_file</attr>
            <attr key="moduleOutput" type="string">0</attr>
            <attr key="runAtStartup" type="bool">true</attr>
        </node>
        <node name="caer" path="/caer/">
            <node name="logger" path="/caer/logger/">
                <attr key="logFile" type="string">/private/caer/caer.log</attr>
                <attr key="logLevel" type="byte">5</attr>
            </node>
            <node name="modules" path="/caer/modules/">
                <attr key="modulesSearchPath" type="string">/private/caer/modules/:/usr/local/share/caer/modules/</attr>
            </node>
            <node name="server" path="/caer/server/">
                <attr key="ipAddress" type="string">127.0.0.1</attr>
                <attr key="portNumber" type="int">4040</attr>
            </node>
        </node>
        <node name="eventVisualizer" path="/eventVisualizer/">
            <attr key="eventHandlerListOptions" type="string">None,Spikes,Input</attr>
            <attr key="rendererListOptions" type="string">Polarity,Frame,IMU_6-axes,2D_Points,Spikes,Spikes_Raster_Plot,ETF4D,Polarity_and_Frames</attr>
            <attr key="eventHandler" type="string">None</attr>
            <attr key="logLevel" type="byte">5</attr>
            <attr key="moduleId" type="short">3</attr>
            <attr key="moduleInput" type="string">5[0]</attr>
            <attr key="moduleLibrary" type="string">caer_visualizer</attr>
            <attr key="renderer" type="string">Polarity</attr>
            <attr key="runAtStartup" type="bool">true</attr>
        </node>
        <node name="eventVisualizerUnfiltered" path="/eventVisualizerUnfiltered/">
            <attr key="eventHandlerListOptions" type="string">None,Spikes,Input</attr>
            <attr key="rendererListOptions" type="string">Polarity,Frame,IMU_6-axes,2D_Points,Spikes,Spikes_Raster_Plot,ETF4D,Polarity_and_Frames</attr>
            <attr key="eventHandler" type="string">None</attr>
            <attr key="logLevel" type="byte">5</attr>
            <attr key="moduleId" type="short">6</attr>
            <attr key="moduleInput" type="string">5[0]</attr>
            <attr key="moduleLibrary" type="string">caer_visualizer</attr>
            <attr key="renderer" type="string">Polarity</attr>
            <attr key="runAtStartup" type="bool">true</attr>
        </node>
        <node name="stat" path="/stat/">
            <attr key="divisionFactor" type="long">1000</attr>
            <attr key="logLevel" type="byte">5</attr>
            <attr key="moduleId" type="short">7</attr>
            <attr key="moduleInput" type="string">5[0]</attr>
            <attr key="moduleLibrary" type="string">caer_statistics</attr>
            <attr key="runAtStartup" type="bool">true</attr>
        </node>
    </node>
</sshs>

breznak, 2018-04-09:
Any progres on replaying from `FileInput`? I managed to store to a file in `FileOutput`, but I'm also failing to read from my caer files. 
Would be nice to have some more detailed doc with example configs for each of the modules.

llongi, 2018-04-09:
I see the problem with the above: <attr key="moduleOutput" type="string">0</attr>
The FileInputModule can open anything in theory, so it doesn't know what it's output is going to be (it depends on the file you open and what you want). You have to set it via 'moduleOutput' attribute manually.
In this case you're telling it to only get events of type 0, which probably don't even exist in the file you've written, so it gets stuck.
If we assume you have a file with events and frames, it'd be: <attr key="moduleOutput" type="string">1,2</attr>
I'm currently looking into the I/O situation (boost::asio rewrite, Windows support, etc.) so I'll think of a better way to handle this.
I mean, part of it is a better GUI to make it easy to select which types, but the InputModules can also try to look at the Source headers and give a better choice of what is possible (ie. if we know this is a file from a DAVIS240C, it can only be types 0-3, from a DVS128 only 0-1, and if we know we recorded only events, only ever type 1, and so on).