inivation/caer #101: a problem by running caer on ubuntu

jianlinlu, 2017-08-15:
when I run caer on ubuntu, this problem occurs:

ERROR: nettcpserver: uv_tcp_bind failed, error -99 (EADDRNOTAVAIL).
Segmentation fault (core dumped)

would someone please tell me what is happening? thanks a lot!

llongi, 2017-08-16:
What version of cAER? What version of Ubuntu?
The error means you have the TCP Server module enabled, and that it tries to bind to some IP:Port that doesn't exist. From the bind() man-page: "A nonexistent interface was requested or the requested address was not local.". So it looks like you have some incorrect value for IP address or port in your caer-config.xml.

jianlinlu, 2017-08-16:
Hi Luca,

Thanks so much for your answer!

I'm using the lastest version of caer and ubuntu 16.04 LTS. I will try to solve the IP problem and let you know what happen later.

Thanks again!

Best wishes
Jianlin Lu

jianlinlu, 2017-08-16:
Hi Luca,

Just a question more.
Would you please tell me the function of the TCP Server module?
Because I think I will disable this module if I can not solve the problem. I'm building a stereo vision system by using two DVS and one Dynapse. So I think the TCP module is not so important for this project. Is that correct?
Thanks!

Jianlin

llongi, 2017-08-16:
Yes, it's only used to send events out over the network, so if you're not doing that you don't need it and can safely disable it. -DOUTPUT_NETWORK=0 on cAER 1.X, or -DENABLE_NETWORK_OUTPUT=0 on cAER 0.9.X.

jianlinlu, 2017-08-16:
Thanks a lot! :-)

jianlinlu, 2017-08-16:
Hi Luca,

I'm so sorry to bother you again.

I have tried to "cmake" caer again and this time I disable TCP by using "-DOUTPUT_NETWORK=0",

but there is still some problems about TCP when I run "caer-bin":

root@lin-ubuntu:/home/lin/caer# caer-bin
2017-08-16 20:19:48 (TZ+0200): NOTICE: Logger: Initialization successful with log-level 5.
2017-08-16 20:19:48 (TZ+0200): NOTICE: Mainloop: 5 modules found.
2017-08-16 20:19:48 (TZ+0200): ERROR: Mainloop: Module 'nettcpserver': No module library for 'caer_output_net_tcp_server' found.
2017-08-16 20:19:48 (TZ+0200): ERROR: Mainloop: Errors in module library loading.
2017-08-16 20:19:48 (TZ+0200): CRITICAL: Mainloop: Failed to start mainloop, please fix the configuration and try again!

I think I should modify "caer-config.xml", right? But I don't know how to do it ... where to modify ...

Thanks again for the help!

best wishes

Jianlin Lu

llongi, 2017-08-16:
Somehow your caer-config.xml specifies that it wants that module. I assume you copied docs/dynapse-config.xml as a starting point. For whatever reason it enables the TCP module with a wrong fixed IP, I've removed it now from the config file, so if you copy docs/dynapse-config.xml again it should work.
We're aware the documentation on several things is sub-par and are working on it. :-)

jianlinlu, 2017-08-16:
Hi Luca,

Thanks so much for the answer!

I just delete the part involves TCP in caer-config, and now it is working. :-)

Thanks again for the help!

best wishes

Jianlin

closed 2017-08-17 by llongi