inivation/caer #102: master version of cAER does not compile

federicohyo, 2017-08-24:
see logs:

federico@zfractal:/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer$ cat make_caer.sh 
rm -rf CMakeCache.txt CMakeFiles
cmake -DDVS128=1  -DEDVS=1 -DDAVIS=1 -DDYNAPSE=1 -DBAFILTER=1 -DFRAMEENHANCER=1 -DCAMERACALIBRATION=1 -DSTATISTICS=1 -DVISUALIZER=1 -DINPUT_FILE=1 -DOUTPUT_FILE=1 -DINPUT_NETWORK=1 -DOUTPUT_NETWORK=1  -DROTATE=1 -DMEDIANTRACKER=1 -DRECTANGULARTRACKER=1 -DDYNAMICRECTANGULARTRACKER=1  -DSPIKEFEATURES=1 -DPOISSONSPIKEGEN=1 -DMEANRATEFILTER=1 -DMEANRATEFILTERDVS=1  -DMONITORNEURONFILTER=1 -DSYNAPSERECONFIG=1 -DFPGASPIKEGEN=1 .

federico@zfractal:/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer$ sh make_caer.sh 
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Check size of size_t
-- Check size of size_t - done
-- Check size of void *
-- Check size of void * - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libcaer>=2.2.0'
--   Found libcaer, version 2.3.0
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Checking for module 'mxml>=2.7'
--   Found mxml, version 2.9
-- Checking for module 'opencv>=3.1'
--   Found opencv, version 3.1.0
-- Checking for module 'libpng>=1.6'
--   
-- Checking for module 'libuv>=1.7.5'
--   Found libuv, version 1.8.0
-- Checking for module 'allegro-5>=5.0.11'
--   
-- Checking for module 'allegro_main-5>=5.0.11'
--   
-- Checking for module 'allegro_primitives-5'
--   No package 'allegro_primitives-5' found
-- Checking for module 'allegro_font-5'
--   No package 'allegro_font-5' found
[... 291 lines cut ...]
 static bool isOutputBeingUsed(int16_t sourceId, int16_t typeId, int16_t afterModuleId, int16_t currModuleId,
             ^
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1257:13: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp: In function ‘void runModules(caerEventPacketContainer)’:
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1444:13: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
 static void runModules(caerEventPacketContainer in) {
             ^
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1444:13: warning: assuming signed overflow does not occur when reducing constant in comparison [-Wstrict-overflow]
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1444:13: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp: In function ‘void parseModuleInput(const string&, std::unordered_map<short int, std::vector<OrderedInput> >&, int16_t, const string&)’:
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:627:13: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
 static void parseModuleInput(const std::string &inputDefinition,
             ^
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:627:13: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp: In function ‘int caerMainloopRunner()’:
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1581:12: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
 static int caerMainloopRunner() {
            ^
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1581:12: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1581:12: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1581:12: warning: assuming signed overflow does not occur when changing X +- C1 cmp C2 to X cmp C2 -+ C1 [-Wstrict-overflow]
[ 20%] Linking CXX executable caer-bin
[ 20%] Built target caer-bin
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

llongi, 2017-08-24:
> /usr/bin/ld: cannot find -lippicv
> collect2: error: ld returned 1 exit status

That suggests some problem with linking. IPP is used by OpenCV so something is wrong with your OpenCV install. There's nothing on the caer side we can do about this.

federicohyo, 2017-08-24:
does this mean that if I remove the -DFRAMEENHANCER=0 -DCAMERACALIBRATION=0  and do not rely on opencv it should succefully compile?

federicohyo, 2017-08-24:
to me it seems that this defition DYNAPSE_MAX_USER_USB_PACKET_SIZE has been removed in libcaer but not it caer  
/inilabs/caer/modules/ini/gen_spikes.c:240:24: error: ‘DYNAPSE_MAX_USER_USB_PACKET_SIZE’ undeclared (first use in this function)^[[0m
   uint32_t bits_chipU0[DYNAPSE_MAX_USER_USB_PACKET_SIZE];

llongi, 2017-08-24:
According to the error yes that would be my assumption. Master definitely still compiles fine.

On August 24, 2017 11:36:15 AM GMT+02:00, Federico Corradi <notifications@github.com> wrote:
>does this mean that if I remove the -DFRAMEENHANCER=0
>-DCAMERACALIBRATION=0  and do not rely on opencv it should succefully
>compile?

-- 
Sent from my Android device with K-9 Mail. Please excuse my brevity.

federicohyo, 2017-08-24:
DYNAPSE_MAX_USER_USB_PACKET_SIZE is not defined... however it is present in caer/modules/ini/gen_spikes.c:240. How can this compile?

llongi, 2017-08-24:
Yes. Use the sfml branch or stable libcaer to fix that one.

federicohyo, 2017-08-24:
so master caer require branch of libcaer...

llongi, 2017-08-25:
Master caer now requires and compiles with libcaer 2.4.0, which is the development version of libcaer. So stable release works with stable release, development version with development version.

federicohyo, 2017-08-28:
I still have problems, here the log:

federico@zfractal:/media/federico/space/inilabs/caer$ sh make_caer.sh 
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Check size of size_t
-- Check size of size_t - done
-- Check size of void *
-- Check size of void * - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libcaer>=2.4.0'
--   Found libcaer, version 2.4.0
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   program_options
-- Checking for module 'mxml>=2.7'
--   Found mxml, version 2.9
-- Checking for module 'libpng>=1.6'
--   
-- Checking for module 'libuv>=1.7.5'
--   Found libuv, version 1.8.0
-- Checking for module 'sfml-graphics>=2.3.0'
--   Found sfml-graphics, version 2.3.2
-- Checking for module 'glew>=1.10.0'
--   Found glew, version 1.13.0
-- Project version is: 1.0.1
-- Compiler is Clang: FALSE
-- Compiler is GCC: TRUE
-- Compiler is IntelCC: FALSE
-- Compiler is MS VisualC: FALSE
-- OS is Unix: TRUE
-- OS is Linux: TRUE
-- OS is MacOS X: FALSE
-- OS is Windows: FALSE
-- System is big-endian: 0
-- Thread support is PThreads: TRUE
-- Thread support is Win32 Threads: FALSE
-- C flags are:  -std=c11 -pedantic -Wall -Wextra -Wunused -Wundef -Wformat=2 -Winit-self -Wuninitialized -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wredundant-decls -Wmissing-declarations -Wstack-protector -Wshadow -Wfloat-equal -Wconversion -Wstrict-overflow=5 -Wdouble-promotion  -Wstrict-prototypes -Wmissing-prototypes -Wnested-externs -Wbad-function-cast -Wjump-misses-init -Wunsuffixed-float-constants
-- CXX flags are:  -std=c++14 -pedantic -Wall -Wextra -Wunused -Wundef -Wformat=2 -Winit-self -Wuninitialized -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wredundant-decls -Wmissing-declarations -Wstack-protector -Wshadow -Wfloat-equal -Wconversion -Wstrict-overflow=5 -Wdouble-promotion 
-- Include directories are: /usr/local/include;/usr/local/include/opencv;/usr/local/include;/usr/include/libusb-1.0;/usr/include;/media/federico/space/inilabs/caer/;/usr/local/include
-- Library directories are: /usr/local/lib;/usr/lib/x86_64-linux-gnu;/media/federico/space/inilabs/caer/;/usr/local/lib
-- Linked C libraries are: caer;-lpthread;m;rt;mxml;pthread
-- Compiled C source files are: base/log.c;base/misc.c;ext/slre/slre.c;ext/sshs/sshs.c;ext/sshs/sshs_helper.c;ext/sshs/sshs_node.c
-- Linked C++ libraries are: caer;-lpthread;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_program_options.so;dl
-- Compiled C++ source files are: base/config.cpp;base/config_server.cpp;base/module.cpp;base/mainloop.cpp
-- Final install bindir is: /usr/local/bin
-- Configuring done
-- Generating done
-- Build files have been written to: /media/federico/space/inilabs/caer

maybe I have the wrong version of sfml graphics? I installed it with 'sudo apt-get install libsfml-dev'

[1;91m/media/federico/space/inilabs/caer/modules/visualizer/visualizer.cpp:747:20: error: ‘class sf::Text’ has no member named ‘setFillColor’[0m
    totalEventsText.setFillColor(sf::Color::White);
                    ^
[1;32m/media/federico/space/inilabs/caer/modules/visualizer/visualizer.cpp:748:71: warning: conversion to ‘float’ from ‘uint32_t {aka unsigned int}’ may alter its value [-Wconversion][0m
    totalEventsText.setPosition(GLOBAL_FONT_SPACING, state->renderSizeY);
                                                                       ^
[1;91m/media/federico/space/inilabs/caer/modules/visualizer/visualizer.cpp:753:20: error: ‘class sf::Text’ has no member named ‘setFillColor’[0m
    validEventsText.setFillColor(sf::Color::White);

see attached log file

[error.txt](https://github.com/inilabs/caer/files/1255419/error.txt)

llongi, 2017-08-28:
Fixed in 31de4d410a1cc72c0674185be79058bde0ac38be.

closed 2017-08-29 by federicohyo