inivation/caer #109: can not compile dynapse application

Thanh-Binh, 2017-09-15:
I tried to compile with option dynapse:
cmake -DDYNAPSE=1 -DSTATISTICS=1  -DVISUALIZER=1 -DINPUT_FILE=1 -DOUTPUT_FILE=1 -DINPUT_NETWORK=1 -DOUTPUT_NETWORK=1 -DMEANRATEFILTER=1 -DSYNAPSERECONFIG=1 -DFPGASPIKEGEN=1 -DPOISSONSPIKEGEN=1 .

 but always have the error like:

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/binh/Neuromorphics/caer/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_0f948/fast"
/usr/bin/make -f CMakeFiles/cmTC_0f948.dir/build.make CMakeFiles/cmTC_0f948.dir/build
make[1]: Entering directory '/home/binh/Neuromorphics/caer/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_0f948.dir/CheckSymbolExists.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_0f948.dir/CheckSymbolExists.c.o   -c /home/binh/Neuromorphics/caer/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_0f948
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_0f948.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_0f948.dir/CheckSymbolExists.c.o  -o cmTC_0f948 -rdynamic 
CMakeFiles/cmTC_0f948.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_0f948.dir/build.make:97: recipe for target 'cmTC_0f948' failed
make[1]: *** [cmTC_0f948] Error 1
make[1]: Leaving directory '/home/binh/Neuromorphics/caer/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_0f948/fast' failed
make: *** [cmTC_0f948/fast] Error 2

llongi, 2017-09-15:
CheckSymbolExists.c.o: In function main': CheckSymbolExists.c:(.text+0x16): undefined reference to 'pthread_create'

that's the actual error and it's not in the cAER code, but in CMake's CheckSymbolExists.c that it uses to try and determine how threads are supported on your system.
What system are you trying to compile this on?
Also please try to remove the CMakeFiles/ directory, as well as the CMakeCache.txt file and try running cmake again.

Thanh-Binh, 2017-09-15:
I tested in ubuntu 14.04 lts. I use pthread in different applications

Thanh-Binh, 2017-09-15:
I always remove Cmakefiles and cmakecache every building

federicohyo, 2017-09-18:
It is definitely an issue with pthread, have you tried installing or reinstalling the package "apt-get install libpthread-stubs0-dev" ?

Thanh-Binh, 2017-09-18:
Yes I did it, see the results of running this script
libpthread-stubs0-dev is already the newest version (0.3-4).

Maybe this lib is compiled by different version of compiler?

llongi, 2017-09-18:
After some searching on the internet this looks like a known problem, sometimes there seem to be strange interaction between libraries, especially Boost, that make this check fail.
A possible solution I found:
pass -DBoost_NO_BOOST_CMAKE=ON to the cmake call
Can you try this and report back if it helps? Thanks!

Thanh-Binh, 2017-09-18:
Thanks. I tried it but have the same problems as before. I found "pthread" lib in my system as follows:

$ find /usr/. -name "*pthread*"
/usr/./lib/x86_64-linux-gnu/libpthread.a
/usr/./lib/x86_64-linux-gnu/pkgconfig/pthread-stubs.pc
/usr/./lib/x86_64-linux-gnu/libgpgme-pthread.so.11
/usr/./lib/x86_64-linux-gnu/libpthread.so
/usr/./lib/x86_64-linux-gnu/libpthread_nonshared.a
/usr/./lib/x86_64-linux-gnu/libgpgme-pthread.so.11.14.0
/usr/./lib/x86_64-linux-gnu/perl/5.22.1/bits/pthreadtypes.ph
/usr/./lib/python3/dist-packages/plainbox/vendor/pyglibc/_pthread_sigmask.py
/usr/./lib/python3/dist-packages/plainbox/vendor/pyglibc/__pycache__/_pthread_sigmask.cpython-35.pyc
/usr/./include/pthread.h
/usr/./include/boost/thread/pthread
/usr/./include/boost/thread/pthread/pthread_mutex_scoped_lock.hpp
/usr/./include/boost/interprocess/sync/posix/pthread_helpers.hpp
/usr/./include/boost/smart_ptr/detail/lwm_pthreads.hpp
/usr/./include/boost/signals2/detail/lwm_pthreads.hpp
/usr/./include/x86_64-linux-gnu/bits/pthreadtypes.h

Thanh-Binh, 2017-09-19:
it works now so that I will close this issues ...

Thanh-Binh, 2017-10-20:
I have downloaded the current cAER and tried to compile it in the same way before, but have got again the error by checking if pthread_create() is available.
If I look at the file CMakeError.log and found the error message here:
```C
Run Build Command:"/usr/bin/make" "cmTC_f9bba/fast"
/usr/bin/make -f CMakeFiles/cmTC_f9bba.dir/build.make CMakeFiles/cmTC_f9bba.dir/build
make[1]: Entering directory '/home/binh/Neuromorphics/current/caer/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_f9bba.dir/CheckFunctionExists.c.o
/usr/bin/cc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_f9bba.dir/CheckFunctionExists.c.o   -c /usr/share/cmake-3.5/Modules/CheckFunctionExists.c
Linking C executable cmTC_f9bba
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_f9bba.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_f9bba.dir/CheckFunctionExists.c.o  -o cmTC_f9bba -rdynamic -lpthreads 
/usr/bin/ld: cannot find -lpthreads
```
It it clear that "pthreads" library does NOT exist in my Ubuntu. I have only "pthread" lib (without the character 's' at the end).

I really do not find out which script generate this false links?
@llongi : Could you please check it? Thanks

llongi, 2017-10-20:
The origin of that check is caer/CMakeLists.txt line 93: FIND_PACKAGE(Threads)
Which execute the code in /usr/share/cmake/Modules/FindThreads.cmake.
This is the standard CMake module to find out what the thread support is on a system.
On most systems, it should succeed with the line:
  CHECK_SYMBOL_EXISTS(pthread_create pthread.h CMAKE_HAVE_LIBC_CREATE)
Somehow on yours it doesn't and goes on to '_check_threads_lib(pthreads ...)'.
This is very suspicious, as any modern Linux system should have the pthread library be part of the libC and be available in pthread.h. For example I have /usr/include/pthread.h with all the expected functions in it. Do you have that file?
At this point it might make sense to ask/open a bug with the CMake people about how this fails on your system, I can't really help you out with that, and I cannot reproduce it on any of the systems I have access to. Or try to reinstall a clean system from scratch.

Thanh-Binh, 2017-10-20:
I can compile any project using pthread_create(), when the links flag is „-lpthread“.
It fails to link if I use „-lpthreads“

Thanh-Binh, 2017-10-23:
It is very strange at my Ubuntu. it is OK after reinstalling libC6, so that I will close this issue, Thanks.

closed 2017-10-23 by Thanh-Binh