inivation/caer #113: cAER seg fault on exit with wrong configuration

federicohyo, 2017-10-03:
GDB trace...

federico@zfractal:/media/federico/space/inilabs/caer$ gdb ./caer-bin
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from ./caer-bin...done.
(gdb) r
Starting program: /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/caer-bin 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
2017-10-03 13:10:22 (TZ+0200): NOTICE: Logger: Initialization successful with log-level 5.
[New Thread 0x7ffff36d0700 (LWP 32651)]
2017-10-03 13:10:23 (TZ+0200): NOTICE: Mainloop: 7 modules found.
2017-10-03 13:10:23 (TZ+0200): ERROR: Mainloop: Found dependency cycle involving multiple streams between modules 'imageGen' (ID 4) and 'optic' (ID 6).
2017-10-03 13:10:23 (TZ+0200): CRITICAL: Mainloop: Failed to start mainloop, please fix the configuration and try again!
^C
Thread 1 "caer-bin" received signal SIGINT, Interrupt.
0x00007ffff799ec1d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
84	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt full
#0  0x00007ffff799ec1d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
No locals.
#1  0x00000000005a3949 in std::this_thread::sleep_for<long, std::ratio<1l, 1l> > (__rtime=...) at /usr/include/c++/5/thread:292
        __s = {__r = 1}
        __ns = {__r = 0}
        __ts = {tv_sec = 1, tv_nsec = 0}
#2  0x0000000000593370 in caerMainloopRun () at /media/federico/space/inilabs/caer/base/mainloop.cpp:319
        result = 1
        shutdown = {__sigaction_handler = {sa_handler = 0x59d3e0 <caerMainloopSignalHandler(int)>, sa_sigaction = 0x59d3e0 <caerMainloopSignalHandler(int)>}, sa_mask = {__val = {
              16386, 0 <repeats 15 times>}}, sa_flags = 0, sa_restorer = 0x889250}
        modulesNode = 0x88ffb0
        modulesBuildDir = {static preferred_separator = 47 '/', m_pathname = "/media/federico/space/inilabs/caer/modules/"}
        modulesDefaultDir = {static preferred_separator = 47 '/', m_pathname = "/usr/local/share/caer/modules/"}
        systemNode = 0x88f6b0
#3  0x00000000005c49d8 in main (argc=1, argv=0x7fffffffda78) at /media/federico/space/inilabs/caer/main.cpp:23
No locals.

federicohyo, 2017-10-03:
==1022== Warning: noted but unhandled ioctl 0x18 with no size/direction hints.
==1022==    This could cause spurious value errors to appear.
==1022==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==1022== Warning: noted but unhandled ioctl 0x42 with no size/direction hints.
==1022==    This could cause spurious value errors to appear.
==1022==    See README_MISSING_SYSCALL_OR_IOCTL for guidance on writing a proper wrapper.
==1022== Thread 1 memcheck-amd64-:
==1022== Invalid read of size 4
==1022==    at 0x128AF834: ??? (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x128AF8AE: google::protobuf::internal::DestroyDefaultRepeatedFields() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x128AEB3A: google::protobuf::ShutdownProtobufLibrary() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x2333F3B8: ??? (in /usr/lib/x86_64-linux-gnu/libmirprotobuf.so.3)
==1022==    by 0x4010DE6: _dl_fini (dl-fini.c:235)
==1022==    by 0x62FFFF7: __run_exit_handlers (exit.c:82)
==1022==    by 0x6300044: exit (exit.c:104)
==1022==    by 0x62E6836: (below main) (libc-start.c:325)
==1022==  Address 0xa78cb7c is 12 bytes inside a block of size 24 free'd
==1022==    at 0x4C2F24B: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==1022==    by 0x23A2E4A6: google::protobuf::internal::DestroyDefaultRepeatedFields() (in /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9.0.1)
==1022==    by 0x128AEB3A: google::protobuf::ShutdownProtobufLibrary() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x2333F3B8: ??? (in /usr/lib/x86_64-linux-gnu/libmirprotobuf.so.3)
==1022==    by 0x4010DE6: _dl_fini (dl-fini.c:235)
==1022==    by 0x62FFFF7: __run_exit_handlers (exit.c:82)
==1022==    by 0x6300044: exit (exit.c:104)
==1022==    by 0x62E6836: (below main) (libc-start.c:325)
==1022==  Block was alloc'd at
==1022==    at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==1022==    by 0x128B69ED: google::protobuf::internal::InitializeDefaultRepeatedFields() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x128AD0C8: ??? (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x40106B9: call_init.part.0 (dl-init.c:72)
==1022==    by 0x40107CA: call_init (dl-init.c:30)
==1022==    by 0x40107CA: _dl_init (dl-init.c:120)
==1022==    by 0x40158E1: dl_open_worker (dl-open.c:575)
==1022==    by 0x4010563: _dl_catch_error (dl-error.c:187)
==1022==    by 0x4014DA8: _dl_open (dl-open.c:660)
==1022==    by 0x5B2AF08: dlopen_doit (dlopen.c:66)
==1022==    by 0x4010563: _dl_catch_error (dl-error.c:187)
==1022==    by 0x5B2B570: _dlerror_run (dlerror.c:163)
==1022==    by 0x5B2AFA0: dlopen@@GLIBC_2.2.5 (dlopen.c:87)
==1022== 
==1022== Invalid read of size 8
==1022==    at 0x128AF85A: ??? (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x128AF8AE: google::protobuf::internal::DestroyDefaultRepeatedFields() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x128AEB3A: google::protobuf::ShutdownProtobufLibrary() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x2333F3B8: ??? (in /usr/lib/x86_64-linux-gnu/libmirprotobuf.so.3)
==1022==    by 0x4010DE6: _dl_fini (dl-fini.c:235)
==1022==    by 0x62FFFF7: __run_exit_handlers (exit.c:82)
==1022==    by 0x6300044: exit (exit.c:104)
==1022==    by 0x62E6836: (below main) (libc-start.c:325)
==1022==  Address 0xa78cb70 is 0 bytes inside a block of size 24 free'd
==1022==    at 0x4C2F24B: operator delete(void*) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==1022==    by 0x23A2E4A6: google::protobuf::internal::DestroyDefaultRepeatedFields() (in /usr/lib/x86_64-linux-gnu/libprotobuf-lite.so.9.0.1)
==1022==    by 0x128AEB3A: google::protobuf::ShutdownProtobufLibrary() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x2333F3B8: ??? (in /usr/lib/x86_64-linux-gnu/libmirprotobuf.so.3)
==1022==    by 0x4010DE6: _dl_fini (dl-fini.c:235)
==1022==    by 0x62FFFF7: __run_exit_handlers (exit.c:82)
==1022==    by 0x6300044: exit (exit.c:104)
==1022==    by 0x62E6836: (below main) (libc-start.c:325)
==1022==  Block was alloc'd at
==1022==    at 0x4C2E0EF: operator new(unsigned long) (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==1022==    by 0x128B69ED: google::protobuf::internal::InitializeDefaultRepeatedFields() (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x128AD0C8: ??? (in /usr/lib/x86_64-linux-gnu/libprotobuf.so.9.0.1)
==1022==    by 0x40106B9: call_init.part.0 (dl-init.c:72)
==1022==    by 0x40107CA: call_init (dl-init.c:30)
==1022==    by 0x40107CA: _dl_init (dl-init.c:120)
==1022==    by 0x40158E1: dl_open_worker (dl-open.c:575)
==1022==    by 0x4010563: _dl_catch_error (dl-error.c:187)
==1022==    by 0x4014DA8: _dl_open (dl-open.c:660)
==1022==    by 0x5B2AF08: dlopen_doit (dlopen.c:66)
==1022==    by 0x4010563: _dl_catch_error (dl-error.c:187)
[... 579 lines cut ...]
==1022==    by 0x4014DA8: _dl_open (dl-open.c:660)
==1022==    by 0x5B2AF08: dlopen_doit (dlopen.c:66)
==1022==    by 0x4010563: _dl_catch_error (dl-error.c:187)
==1022==    by 0x5B2B570: _dlerror_run (dlerror.c:163)
==1022==    by 0x5B2AFA0: dlopen@@GLIBC_2.2.5 (dlopen.c:87)
==1022== 
==1022== 
==1022== HEAP SUMMARY:
==1022==     in use at exit: 378,119,674 bytes in 274,244 blocks
[... cut ...]
==1022==                         length64           : 5,376 bytes in 87 blocks
==1022==                         newarray           : 2,096 bytes in 51 blocks
==1022==         suppressed: 0 bytes in 0 blocks
==1022== Rerun with --leak-check=full to see details of leaked memory
==1022== 
==1022== For counts of detected and suppressed errors, rerun with: -v
==1022== Use --track-origins=yes to see where uninitialised values come from
==1022== ERROR SUMMARY: 27 errors from 23 contexts (suppressed: 0 from 0)

llongi, 2017-10-03:
Seems similar to https://github.com/google/protobuf/issues/206 and this looks very relevant:
https://bugs.launchpad.net/mir/+bug/1391976
Looks like a bug in the Ubuntu Mir display server or Protobuf itself, related to how you cannot load/unload them safely. Comment #9 there is pretty much what we are doing. They say they fixed it, but then there is this:
https://bugs.launchpad.net/mir/+bug/1619616
That one seems fixed in libprotobuf-dev 3.0.0 but only the Ubuntu version.
If you have installed Protobuf manually, or it was built-in as part of something else (Caffe?), then it doesn't have that fix and crashes.
There's nothing we can do in cAER here, it's an indirect external library bug.

closed 2017-10-03 by llongi