inivation/caer #121: deadlock when chaining multiple modules

federicohyo, 2017-10-23:
if a module is defined as changing the input stream as for example an input stream of events.

	{ .type = POLARITY_EVENT, .number = 1, .readOnly = false }

and if the output of this module is used as input for multiple modules, then a deadlock occurs.

Here the trace of where the deadlock happens:

2017-10-23 11:03:32 (TZ+0200): NOTICE: Mainloop: 12 modules found.
^C
Thread 1 "caer-bin" received signal SIGINT, Interrupt.
0x00000000005fc5cc in std::vector<DependencyLink, std::allocator<DependencyLink> >::begin (this=0xaae828) at /usr/include/c++/5/bits/stl_vector.h:548
548	      { return iterator(this->_M_impl._M_start); }
(gdb) bt full
#0  0x00000000005fc5cc in std::vector<DependencyLink, std::allocator<DependencyLink> >::begin (this=0xaae828) at /usr/include/c++/5/bits/stl_vector.h:548
No locals.
#1  0x00000000005edfb7 in IDExistsInDependencyTree (root=0xaae810, searchId=9, directionUp=false)
    at /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:926
        depLink = @0x8: <error reading variable>
        __for_range = std::vector of length 5, capacity 8 = {{id = -1, next = std::shared_ptr (count 1, weak 0) 0xad39c0}, {id = -1, 
            next = std::shared_ptr (count 1, weak 0) 0xaae8d0}, {id = 2, next = std::shared_ptr (count 1, weak 0) 0xaae860}, {id = 4, 
            next = std::shared_ptr (empty) 0x0}, {id = 7, next = std::shared_ptr (empty) 0x0}}
        __for_begin = {id = -12480, next = std::shared_ptr (count -326940589, weak 2106148887) 0x13dc430}
        __for_end = {id = -15288, next = std::shared_ptr (count 11432416, weak -1) 0x13dc448}
#2  0x00000000005ee051 in IDExistsInDependencyTree (root=0xad3920, searchId=9, directionUp=false)
    at /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:934
        found = {first = 0x7fffffffcf90, second = 0x7fffffffcf40}
        depLink = @0x13dc430: {id = 1, next = std::shared_ptr (count 1, weak 0) 0xaae810}
        __for_range = std::vector of length 1, capacity 1 = {{id = 1, next = std::shared_ptr (count 1, weak 0) 0xaae810}}
        __for_begin = {id = 1, next = std::shared_ptr (count 1, weak 0) 0xaae810}
        __for_end = {id = 24930, next = <error reading variable: Cannot access memory at address 0x89>}
#3  0x00000000005ef394 in mergeActiveStreamDeps () at /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1192
        stream = @0xae71e0: {sourceId = 9, typeId = 9, isProcessor = true, users = std::vector of length 1, capacity 1 = {10}, 
          dependencies = std::shared_ptr (count 1, weak 0) 0xad3840}
        streamSourceId = 9
        originExists = {first = 0x0, second = 0x0}
        mergeResult = std::shared_ptr (count 1, weak 0) 0xad3920
        processorStreams = std::queue wrapping: std::deque with 0 elements
        finalModuleOrder = std::vector of length -70368740985263, capacity -70368744171408 = {14624, 173, 0, 0, 14608, 173, 0, 0, 0, 0, -1, -1, 14504, 173, 0, 
          0, 14688, 173, 0, 0, 8, 0, 0, 0, -6216, 170, 0, 0, -6672, 170, 0, 0, -6160, 170, 0, 0, 14728, 173, 0, 0, -6216, 170, 0, 0, -6672, 170, 0, 0, -6160, 
          170, 0, 0, 14728, 173, 0, 0, 29152, 174, 0, 0, 11984, 173, 0, 0, 14848, 174, 0, 0, 14848, 174, 0, 0, 15360, 174, 0, 0, 4872, 172, 0, 0, 14848, 174, 
          0, 0, 14848, 174, 0, 0, 15360, 174, 0, 0, 4872, 172, 0, 0, -12176, -1, 32767, 0, -15616, -18142, -3173, 7000, 9, 0, 0, 0, 23802, 173, 0, 0, -10288, 
          -1, 32767, 0, 13032, 95, 0, 0, -12176, -1, 32767, 0, -13376, 172, 257, 10, 8, 1, 32767, 0, 14320, 88, 0, 0, 29200, 174, 0, 0, 0, 0, 0, 0, 0, 0, 0, 
          0, -27332, -2465, 32767, 0, -1, -1, 0, 0, 0, 0, 0, 0, 29200, 174, 0, 0, -27332, -2465, 32767, 0, 29200, 174, 0, 0, 13, 0, 0, 0, 0, 0, 0, 0, 13, 0, 
          0, 0, -9616, -1, 32767, 0, -13392, 172, 0, 0, 20688, 145, 0, 0, 31168, 140, 0, 0...}
---Type <return> to continue, or q <return> to quit---
        currLevel = std::vector of length 0, capacity 0
        nextLevel = std::vector of length -2, capacity -538289956 = {0x0, 0x7fffffffffff, 0x0, 0x13dc430, 0x13dc448, 0x13dc448, 0xaae5f0, 0x51, 0x0, 
          0x7ffff6939b78 <main_arena+88>, 0x5c197c
     <std::_Function_base::_Base_manager<std::__detail::_CharMatcher<std::__cxx11::regex_traits<char>, false, false> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation)>, 0xaae5f0, 0xae3a00, 0xaae5f0, 0xae3a00, 0xaae5f0, 0x7fff77ffc4e0, 0x51, 
          0x65af78 <vtable for std::_Sp_counted_ptr_inplace<DependencyNode, std::allocator<DependencyNode>, (__gnu_cxx::_Lock_policy)2>+16>, 0x100000001, 0x2, 
          0xffff, 0xaae810, 0x0, 0x0, 0x0, 0x0, 0x51, 
          0x65af78 <vtable for std::_Sp_counted_ptr_inplace<DependencyNode, std::allocator<DependencyNode>, (__gnu_cxx::_Lock_policy)2>+16>, 0x100000001, 0x3, 
          0x7ffff693ffff, 0xad39c0, 0xad2e90, 0xad2ec0, 0xad2ec0, 0xab70c0, 0x51, 
          0x65af78 <vtable for std::_Sp_counted_ptr_inplace<DependencyNode, std::allocator<DependencyNode>, (__gnu_cxx::_Lock_policy)2>+16>, 0x100000001, 0x3, 
          0x373738633464000c, 0xaae8d0, 0xad2f40, 0xad2f58, 0xad2f58, 0x73656c75646f4d65, 0x21, 0x0, 0x7ffff6939ba8 <main_arena+136>, 0x300, 0x21, 0xac0006, 
          0xac12b0, 0xac12a0, 0x141, 0xad3af0, 0xd, 0x745374757074756f, 0x736d616572, 0xad3c20, 0x36, 0x50, 0x0, 0xad3570, 0x0, 0x0, 0xad3ec0, 0xad3ec0, 
          0xad3ec0, 0x1, 0x0, 0x0, 0xacfd10, 0xacfd10, 0xacfd10, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x1, 0x0, 
          0x0, 0x61, 0x6f6d2f726561632f, 0x61632f73656c7564, 0x7475706e695f7265, 0x7063745f74656e5f, 0x2f746e65696c635f, 0x745374757074756f, 0x2f736d616572, 
          0x0, 0x0, 0x0, 0x0, 0x51, 0x1, 0xad35c0, 0x0, 0x0, 0xad3cb0, 0xd, 0x745374757074756f, 0x736d616572, 0xad3ae0, 0xb1, 0x7fff00000000, 0xacc860, 0x0, 
          0x0, 0xad3d00, 0x4, 0x65707974, 0x0, 0x1, 0x40, 0x5, 0xad3d38, 0xc, 0x7420656c75646f4d, 0x2e657079, 0x7, 0x0, 0xad3d68, 0x9, 0x4f535345434f5250, 
          0x52, 0x141, 0xad3d90, 0x1, 0x30, 0x0, 0x92e9c0, 0x38, 0x6c, 0x0, 0xad3ae0, 0x0, 0x0, 0x0, 0xad3dd0, 0xad3dd0, 0x0, 0x0, 0x0, 0xacfdc0, 0xacfdc0, 
          0xacfdc0, 0x1, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x1, 0x0, 0x0, 0x51, 0x1, 0xad3b30, 0x0, 0x0, 0xad3ef0, 
          0x1, 0x7fffffff0030, 0x7fffffffd160, 0xad3d80, 0x81, 0x6f6d2f726561632f, 0x61632f73656c7564, 0x757074756f5f7265, 0x63745f74656e5f74, 
          0x746e65696c635f70, 0x74537475706e692f, 0x2f302f736d616572, 0x7fffffffd000, 0x656c75646f6d2f00, 0x7fffffffd0f0...}
#4  0x00000000005f32e8 in caerMainloopRunner () at /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/base/mainloop.cpp:1905
        modulesSize = 13
        modules = 0xaccbb0
        inputModules = std::vector of length 1, capacity 1 = {
          {<std::_Reference_wrapper_base<ModuleInfo>> = {<std::_Reference_wrapper_base_impl<false, false, ModuleInfo>> = {<std::_Weak_result_type<ModuleInfo>> = {<std::_Weak_result_type_impl<ModuleInfo>> = {<std::_Maybe_get_result_type<ModuleInfo, void>> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, _M_data = 0x30e5890}}
[... 6 lines cut ...]
          {<std::_Reference_wrapper_base<ModuleInfo>> = {<std::_Reference_wrapper_base_impl<false, false, ModuleInfo>> = {<std::_Weak_result_type<ModuleInfo>> = {<std::_Weak_result_type_impl<ModuleInfo>> = {<std::_Maybe_get_result_type<ModuleInfo, void>> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, _M_data = 0xace720}, 
          {<std::_Reference_wrapper_base<ModuleInfo>> = {<std::_Reference_wrapper_base_impl<false, false, ModuleInfo>> = {<std::_Weak_result_type<ModuleInfo>> = {<std::_Weak_result_type_impl<ModuleInfo>> = {<std::_Maybe_get_result_type<ModuleInfo, void>> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, _M_data = 0xace610}}
        processorModules = std::vector of length 5, capacity 8 = {
          {<std::_Reference_wrapper_base<ModuleInfo>> = {<std::_Reference_wrapper_base_impl<false, false, ModuleInfo>> = {<std::_Weak_result_type<ModuleInfo>> = {<std::_Weak_result_type_impl<ModuleInfo>> = {<std::_Maybe_get_result_type<ModuleInfo, void>> = {<No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, <No data fields>}, _M_data = 0xaa75c0}, 
[... cut ...]
        modulesNode = 0x8f4ae0
        modulesBuildDir = {static preferred_separator = 47 '/', m_pathname = "/media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/modules/"}
        modulesDefaultDir = {static preferred_separator = 47 '/', m_pathname = "/usr/local/share/caer/modules/"}
        systemNode = 0x8f4500
#6  0x0000000000651075 in main (argc=1, argv=0x7fffffffda78) at /media/federico/e247eb2e-273d-4a18-895a-6dd4c877d551/inilabs/caer/main.cpp:23
No locals.

[image]
[image]
[image]

llongi, 2017-10-23:
Confirmed, bug is in Mainloop module merge&order part. Attached a simple XML config file that exhibits this right now.
[test.txt](https://github.com/inilabs/caer/files/1407002/test.txt)

llongi, 2017-10-23:
I've confined the bug the TODO in commit 7fd8460d167e08125607d32d5230c90f2797d935. It's an infinite recursion, not a dead-lock, so bug title updated.
Will continue analysis tomorrow.

llongi, 2017-10-24:
Fixed in 05f1933573f1adf4c734f3468fff372798a2226b.

closed 2017-10-24 by llongi