inivation/caer #124: ring-buffer full?

Thanh-Binh, 2017-10-25:
I run today the new version with davis240b, I have 
./caer-bin-davis-vis -c docs/davis-configB.xml 
2017-10-25 09:53:06 (TZ+0200): NOTICE: Logger: Initialization successful with log-level 5.
2017-10-25 09:53:06 (TZ+0200): NOTICE: Mainloop: 4 modules found.
2017-10-25 09:53:07 (TZ+0200): NOTICE: DAVIS ID-1 SN-04010098 [1:9]: Dropped EventPacket Container because ring-buffer full!

Any idea? Thanks

Thanh-Binh, 2017-10-25:
After reflashing davis240 and 640 as Federico mailed, I am still not able to see anything by visualizing...
Even if I use network client to receiving data, I do not get any data...
Do you have this problem? Solutions?
[davis-configC (copy).xml.txt](https://github.com/inilabs/caer/files/1414241/davis-configC.copy.xml.txt)

llongi, 2017-10-25:
The buffer full message right at the start for a few seconds is normal,
during the time it takes for the visualizer to initialize it blocks
processing and you get that message, as soon as the visualizer starts
the message goes away.

The real problem was the faulty config file.
Closing this bug.

closed 2017-10-25 by llongi