inivation/caer #127: Netparser

kimyo6835, 2017-11-28:
Hi,

I'm trying to make connections with netparser module.
What I want to do is to connect 5 neurons in U1C0 to 6 neurons in U0C0.
( So all 5 neurons are connected to all 6 neurons )
But there is error saying SRAM size limit.

First, I tried this and it worked: Connecting 5 neurons in U1C0 to one neuron in U0C0.

U01-C00-N000-3-10->U00-C00-N000
U01-C00-N001-3-10->U00-C00-N000
U01-C00-N002-3-10->U00-C00-N000
U01-C00-N003-3-10->U00-C00-N000
U01-C00-N004-3-10->U00-C00-N000

But , this didn't work (SRAM size limit)....

U01-C00-N000-3-10->U00-C00-N000
U01-C00-N001-3-10->U00-C00-N000
U01-C00-N002-3-10->U00-C00-N000
U01-C00-N003-3-10->U00-C00-N000
U01-C00-N004-3-10->U00-C00-N000
U01-C00-N000-3-10->U00-C00-N001
U01-C00-N001-3-10->U00-C00-N001
U01-C00-N002-3-10->U00-C00-N001
U01-C00-N003-3-10->U00-C00-N001
U01-C00-N004-3-10->U00-C00-N001
U01-C00-N000-3-10->U00-C00-N002
U01-C00-N001-3-10->U00-C00-N002
U01-C00-N002-3-10->U00-C00-N002
U01-C00-N003-3-10->U00-C00-N002
U01-C00-N004-3-10->U00-C00-N002
U01-C00-N000-3-10->U00-C00-N003
U01-C00-N001-3-10->U00-C00-N003
U01-C00-N002-3-10->U00-C00-N003
U01-C00-N003-3-10->U00-C00-N003
U01-C00-N004-3-10->U00-C00-N003
U01-C00-N000-3-10->U00-C00-N004
U01-C00-N001-3-10->U00-C00-N004
U01-C00-N002-3-10->U00-C00-N004
U01-C00-N003-3-10->U00-C00-N004
U01-C00-N004-3-10->U00-C00-N004
U01-C00-N000-3-10->U00-C00-N005
U01-C00-N001-3-10->U00-C00-N005
U01-C00-N002-3-10->U00-C00-N005
U01-C00-N003-3-10->U00-C00-N005
U01-C00-N004-3-10->U00-C00-N005

federicohyo, 2017-11-30:
Yes, I can confirm the bug. We are now working on a better version of the netparser. 
I will update you when this will be fixed.

sanfans, 2017-12-10:
The issue has been resolved. Now the module is able to intelligently write SRAM cells when is really required. In this way the number of possible connections has been improved significantly. At the end of the documentation (DOCUMENTATION.html in netparser/documentation folder) there are informations about module limitations and hints on how to use it.

llongi, 2017-12-11:
OK great, closing this.

closed 2017-12-11 by llongi