inivation/caer #128: Opening cAER recordings

NatINI, 2017-11-29:
Hi,

I have been trying to open one of my cAER recordings and I am getting the following "warning" message:
WARNING: 10-FileInput: Impossible to determine display sizes from Source information/string. Falling back to 640x480 (VGA).
Segmentation fault: 11

I have already enabled the 10-FileInput and entered the path of my file.

Many thanks for your help.

llongi, 2017-11-29:
When was this recording done? In theory anything done in the last 1.5-2 years should have that information in the file.
Can you open the file and post here the few lines of text at the start?

NatINI, 2017-12-04:
The recording was done 2 months ago so not old. You would like the 1st few lines of the .aedat file? I can maybe sent you that file itself directly.

llongi, 2017-12-04:
Sure, just send it to me and I can take a look.

NatINI, 2017-12-04:
Thanks Luca!​
 OD1-2017_11_21_12_33_27.aedat
<https://drive.google.com/a/tcd.ie/file/d/16KOrwTKABMUdLM7MbOcVv8H04_XcSBC5/view?usp=drive_web>
​

llongi, 2017-12-04:
There are two issues at work here:

1) the Source line in the file is wrong, line 3 of the file:

`#Source 1: DYNAPSE_CONFIG_DYNAPSE_U2`

should read only:

`#Source 1: DYNAPSE`

Please fix that in the file manually. I've verified this doesn't happen with the latest libcaer/cAER anymore, must have been an old bug.

2) the file input module was missing the case for the DYNAPSE, so it didn't know what to do with it and fell back to VGA. This is fixed in commit f9ba93ee141402e04ff5b7ca5a7cf27206e81617.

closed 2017-12-04 by llongi