inivation/caer #133: How to change parameters of one module by another module

ldcyx, 2018-01-03:


ldcyx, 2018-01-04:
2018-01-04 11:57:35 (TZ+0100): EMERGENCY: SSHS: sshsNodePutAttribute(): attribute 'doStimPrimitiveCam' (type 'bool'): attribute doesn't exist, you must create it first.

llongi, 2018-05-11:
If you know either the module ID or the module's name, you can always get a reference to its config node (sshsNode), which is what you should be manipulating to change things. BUT if that configuration doesn't exist then you get the above message.
To make sure the configuration exists, the other module should be using the moduleConfigInit function (see modules/dvsnoisefilter/dvsnoisefilter.c caerDVSNoiseFilterConfigInit() for an example). The moduleConfigInit function is called as soon as a module is added to the main execution loop and can create configuration right away, before the module's initialization or run functions are even called.
Only in very rare cases (for example DAVIS input module) this cannot be done because the configuration depends on information that can only be discovered after full initialization. In that case if you have a data dependency on that module, your own module won't start before the source is fully initialized, so this problem is not possible anymore even in that case.
Fundamentally now only IF the module you want to change does NOT use moduleConfigInit function AND your module does NOT depend in any way on that other module, you can get the 'attribute doesn't exist' error. Since we don't support modules with no data dependency, this can be considered fixed.

closed 2018-05-11 by llongi