inivation/caer #139: DYNAP-SE user guide link to caer software doesn't help

amwhatley, 2018-02-09:
Starting from here: https://ai-ctx.com/support/  and trying to follow the Dynapse getting started instructions in the Dynapse user guide, clicking the link and the end of point 4. "Install cAER, following the instruction in the user guide." sends me here: https://inilabs.com/support/software/caer/
but this does not help as there are no installation instructions at that address, or even any link to where I might be able to download the software from.

clauniel, 2018-02-12:
Changed to point to the repo readme.

amwhatley, 2018-02-12:
Now when I click on the link, unless I middle-click to open it in a new tab, I get the following not very friendly and slightly alarming message:

Blocked by Content Security Policy

This page has a content security policy that prevents it from being loaded in this way.

Firefox prevented this page from loading in this way because the page has a content security policy that disallows it.

llongi, 2018-05-11:
Pointing at README is fine for now. Indeed firefox doesn't load it.
ai-CTX should fix the link and update the instructions in that user guide, it still refers to copying main_dynapse.c to main.c which is ancient cAER from Q1 2017.
Nothing to do for iniVation/cAER here though, so closing.

closed 2018-05-11 by llongi