inivation/caer #143: Issue Installing caer

krebsm249, 2018-03-14:
I'm currently working on a JETSON TX2 trying to get an AEDAT file from the Davis240c. I can get the libcaer davis-sample to run, but i'm having issues with the caer. I'm new to using this, so any help would be appreciated.

cmake -DDAVIS=1 -DINPUT_FILE=1
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   iostreams
--   program_options
--   regex
-- Checking for module 'opencv>=3.1'
--   
CMake Error at /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:367 (message):
  A required package was not found
Call Stack (most recent call first):
  /usr/share/cmake-3.5/Modules/FindPkgConfig.cmake:532 (_pkg_check_modules_internal)
  modules/cameracalibration/CMakeLists.txt:7 (PKG_CHECK_MODULES)

-- Configuring incomplete, errors occurred!
See also "/home/nvidia/Downloads/caer-master/CMakeFiles/CMakeOutput.log".
See also "/home/nvidia/Downloads/caer-master/CMakeFiles/CMakeError.log".

clauniel, 2018-03-16:
Do you have openCV 3.1 or higher installed?

krebsm249, 2018-03-16:
Clauniel,

I do not have opencv 3.1 because the jetson only supports tegra 2.4 for opencv. Also it says it's optional?

clauniel, 2018-03-18:
It is required for the camera calibration module.

llongi, 2018-03-18:
Cameracalibration module needs OpenCV, this is documented in the README and checked during build. Please install it.

closed 2018-03-18 by llongi