inivation/caer #146: Error when running Make

krebsm249, 2018-04-02:
I'm trying to build caer on a Jetson TX2 running Ubuntu 16.04. I run cmake with everything enabled and it works, but when I run make I get errors. 

Errors: 
'shared_ptr' in namespace 'std' does not name a template type

Also get errors with incomplete type:
invalid use of incomplete type 'class std:: reference_wrapper<ModuleInfo>'

llongi, 2018-04-03:
std::shared_ptr is part of C++11, which we do use and specify as a dependency. We check that GCC has the right version too.
Are you by any chance changing anything in the CMake files or passing some custom CFLAGS?
More error output would also be useful.

llongi, 2018-04-03:
I've added some more include directives, please test the latest git.

krebsm249, 2018-04-03:
I got it to build, but caer-bin gives me errors. I am using the config file you uploaded for aedat. https://github.com/inilabs/caer/blob/master/docs/davis-record-file.xml 

I updated the directory to /home/nvidia/Documents

The error is ...

nvidia@tegra-ubuntu:/usr/local/bin$ sudo caer-bin -c '/home/nvidia/Downloads/cr-master/docs/davis-config.xml' 
2018-04-03 16:56:17 (TZ+0000): NOTICE: Logger: Initialization successful with log-level 5.
2018-04-03 16:56:17 (TZ+0000): ERROR: Mainloop: No valid modules configuration found.
2018-04-03 16:56:17 (TZ+0000): CRITICAL: Mainloop: Failed to start mainloop, please fix the configuration and try again!

krebsm249, 2018-04-03:
I was able to run the davis-config.xml and it works, but i cant get the output module to work for aedat