inivation/caer #147: Config for capturing APS, Accelerometer, and AEDAT?

krebsm249, 2018-04-05:
Is there modules or a config file out there to capture APS, Accelerometer and AEDAT for a davis 240c

llongi, 2018-04-05:
Hi, yes the output file module can do this.
You can use the 'docs/davis-record-file.xml' config as a starting point, you then just need to modify the 'moduleInput' attribute (<attr key="moduleInput" type="string">1[1a2]</attr>) so that it also gets eventy of type frame (2) and IMU (3).
An example might look like this:
<attr key="moduleInput" type="string">1[1a2,2,3]</attr>

krebsm249, 2018-04-05:
I was able to get the Frames, but still no IMU data. Also for some reason it says the Frame rate is 0hz

llongi, 2018-04-05:
How do you verify that there is no IMU data? And where does it tell you the frame-rate is 0Hz?

krebsm249, 2018-04-05:
I am looking at the data in jaerviewer on another computer

llongi, 2018-04-05:
Ah, the 0Hz framerate thing is as far as I remember a bug in jAER, due to how things are timestamped when extracting the AEDAT3.X frame. You may want to open an issue in the jAER repository about that.
Regarding IMU data, the jAER AEDAT3.X parser does not support it at the moment, only Polarity Events and Frames.

closed 2018-04-05 by llongi