inivation/caer #152: Setting sshsNode parameters of other modules

ldcyx, 2018-04-30:
When trying to set parameters of a module from another module like so:

	sshsNode spikeNode = sshsGetNode(sshsGetGlobal(), "/poissonSpikeGen/");
	sshsNodePutBool(spikeNode, "Run", false); 

This value is not propagated through. Using a debugger I can verify that the 

    caerPoissonSpikeGenModuleConfig

 function is called, but 

    bool newRun = sshsNodeGetBool(moduleData->moduleNode, "Run");

Is true.

llongi, 2018-04-30:
Can you explain in more detail what is called where? Where are you checking the Run value?
What you're doing should in theory work, if 'caerPoissonSpikeGenModuleConfig' is called then it means the value did change, so the 'newRun' variable in that function should be false. Are you saying is stays true?

llongi, 2018-04-30:
No that looks fine, indeed when you change the parameter to false you see the config function of the spikegen module run in the debugger.
Now my question is, where do you see the Run variable still being true? In the SpikeGen module's config function, or in your own module?

ldcyx, 2018-04-30:
In my module, when I call this function:

	sshsNodeGetBool(spikeNode, "Run"); 

It always returns true after I do:

 	sshsNodePutBool(spikeNode, "Run", false);